pikvm/pikvm #933: Add 'enable' parameter to org mouse and kbd device configuration to allow them to be disabled at boot

interi-mhickman, 2023-02-03:
**Is your feature request related to a problem? Please describe.**
Certain devices we are using with the PiKVM (ADLINK COM Express Type 6 motherboards with AMI BIOS) do not recognize the Mass Storage Device if, when they power on, the PiKVM has both MSD and HID devices enabled. If only the MSD device is enabled, it will appear in the BIOS boot device list. The HID devices can be enabled _after_ immediately after the motherboard powers on, and everything works as expected.

We primarily use the PiKVM in automated test environment to mount media and only occasionally use the KVM features to interact with the console, so we want the PiKVM to boot in a configuration that always works as an MSD for the systems under test.

The problem we would like solved is that there is no way to configure the mouse/kbd HID devices in the override.yaml to be disabled when the PiKVM starts. 

**Describe the solution you'd like**
Add an 'enabled' parameter (as some other devices have) to kvmd:hid or kvmd:hid:mouse and kvmd:hid:keyboard so we can set it to 'false' in the override.yaml and have these devices disabled at boot-up of the PiKVM. Then we will be able to use kvmd-otgconf -e hid.usb[01] to enable them as needed.

**Describe alternatives you've considered**
Plan is to create a systemd unit file that runs at boot and executes kvmd-otgconf -d hid.usb[01].

**Additional context**
This feature request was discussed in the Discord #support channel on 2/3/2023 https://discord.com/channels/580094191938437144/896216359963332628/1071100359109513256. Thank you Arch1mede and mdevaev for your assistance.

interi-mhickman, 2023-02-03:
The following shell script and systemd drop-in work perfectly as an alternative, so this feature request is not critical to our use case, although I do think it makes sense to add to the product.

[root@pikvm ~]# cat /usr/local/bin/disable_hid
#!/usr/bin/env sh

# disable the mouse and keyboard otg devices
/sbin/kvmd-otgconf -d hid.usb0
/sbin/kvmd-otgconf -d hid.usb1

[root@pikvm ~]# cat /etc/systemd/system/kvmd.service.d/disable_hid.conf
[Unit]
After=kvmd.service

[Service]
ExecStartPre=/usr/local/bin/disable_hid

mdevaev, 2023-02-25:
Hello. It will be available in the next release. Use the following in `override.yaml`:
```yaml
otg:
    devices:
        hid:
            keyboard:
                start: false
            mouse:
                start: false
```

mdevaev, 2023-02-25:
PS: Maybe tomorrows or monday.

closed 2023-02-25 by mdevaev