pikvm/pikvm #942: Add support for a 2nd Videostream

PSc-secunet, 2023-02-10:
**Is your feature request related to a problem? Please describe.**
It would be very useful to have the option to configure a 2nd Video stream for an USB Webcam. We sometimes have issues with Notebooks, that dont setup the external Display at BIOS Level. In this case it would be usefull to point a Webcam at the Displsay and enable a 2nd VideoStream in the webUI

**Describe the solution you'd like**
Provide a way to use 2 VideoStream Sources in parallel or an easy way to switch between them in the WebUI

mdevaev, 2023-02-10:
Low priority, but it will be implemented in the future.

mdevaev, 2023-08-10:
I'm closing this issue becase it turned out that we already have a task about it: #894

closed 2023-08-10 by mdevaev