pikvm/pikvm #944: Simple DNS based firewall for PIKVM

etkaar, 2023-02-11:
As far as I know, PIKVM does not yet have a firewall management, at least if someone wants to restrict access to PIKVM to its dynamic IPv4 or IPv6 address. For such use cases I developed [nftm (Nftables Managing Script)](https://github.com/etkaar/nftm) which is a lightweight managing script based on nftables and heavily tested on Debian 10/11. It offers a simple preset + whitelist/blacklist configuration scheme and supports DynDNS as one can also hostnames instead of single IP addresses or subnets.

Is this something which could be useful for PIKVM? As _nftm_ is POSIX-compliant, it should run on Arch Linux.

arch1mede, 2023-02-15:
I am going to put in my 6 cents, if your firewall needs a iptable to reference as most do, you would need to modify this to point to this PST partition, would also need to modify this to be smart and detect how much space is left and make modifications to that same table so it does not get bigger than the available space as other things are occupying that same space.

Besides, I would think the TOTP that was recently implemented would help with who can access, also I think this should be up to the end user, this is unrelated to PiKVM and more arch linux. Of course the dev might already have a solution for this in the works.

etkaar, 2023-02-15:
The thing with the PST partition I don't really understand. Why is that required? The _nftm_ firewall could be either included in PIKVM OS itself or manually installed afterwards.

mdevaev, 2023-02-25:
PST is necessary because the entire root file system is in RO mode.

As for this task itself, I'm not inclined to make quick decisions and apply it in this way. If we are going to make a firewall, then it should have integration with the API and the rest of the PiKVM services. We already have a task about firewall, so I'm closing this as a duplicate of (#201) until better times.

closed 2023-02-25 by mdevaev