pikvm/pikvm #945: Emulated serial adapter

shawly, 2023-02-12:
**Is your feature request related to a problem? Please describe.**

I only have one NVIDIA GPU in my server which I use with vgpu-kvm drivers, since the latest drivers basically disable the virtual terminal, I cannot see any output of my server on the KVM console. If a kernel panic happens, I cannot see the output because the latest drivers disable the framebuffer, so half way through booting my server will *appear* to be frozen.
Currently the only solution is to get a second GPU or a CPU with an integrated GPU or access the host through the motherboards serial port with a serial to USB adapter and minicom.

My case is a very niche case, but this feature would also help people that only have a single GPU in their system that use VFIO for GPU passthrough. If the GPU of their host is passed through to a virtual machine, PiKVM becomes useless if the hosts network is down for example and they have no way to actually shut down their virtual machine and re-bind their GPU to the host for debugging.

**Describe the solution you'd like**

I would like to suggest adding the support of emulating a USB serial adapter through OTG. Since emulating hid and mass storage works nicely, emulating a serial adapter would surely work as well, right?

I know PiKVM supports OTG serial to access PiKVM through the host, just for clarification, I want to do the opposite!

**Describe alternatives you've considered**

Connecting a USB to serial adapter to the Pi and the hosts serial port and then accessing the host through something like minicom works as well. But its not as nice as having it integrated into PiKVM since it blocks a USB port.

Getting a second GPU would also work, but is expensive and for me this wouldn't work as all my PCIe slots are used already.
Getting a CPU with an integrated GPU so I can connect that to PiKVM would work as well, but its also expensive and feels so unnecessary just for getting an iGPU.

**Additional context**

It could look like this:
[image]

Or integrated into the KVM page:

[image]

(these are just mockups)

Also technically, if you supported the OTG emulated console AND external serial to USB adapters, one could add multiple hosts through serial which is waaay cheaper than getting an HDMI KVM switch. Only downside is that BIOS access is not possible since most consumer bioses don't support serial output. Though some server motherboards do support serial console redirection.

mdevaev, 2023-02-25:
Hi. This is already being implemented now. You will need to repeat [this instruction](https://docs.pikvm.org/usb_serial/), but with minor changes.

1. On the PiKVM side: execute only step 1.
2. Execute all steps but for `/dev/ttyACM0` instead of `ttyGS0`. Perhaps PiKVM will not be ttyACM0, then find out what it is.
3. Use `screen /dev/ttyGC0 115200` to access to your server.

mdevaev, 2023-02-28:
Good! :)

As for your request - we have a task about it, but it is a low priority. For now, you can use the web terminal.

#635

closed 2023-02-28 by mdevaev