pikvm/pikvm #946: Better support for removable devices (e.g. USB GPIO boards)

zappanaut, 2023-02-16:
**Is your feature request related to a problem? Please describe.**

I am using a RP2040 as a GPIO extension for ATX control on multiple PCs which is added as a GPIO device in the `override.yaml` file.
Unfortunately a removable device lets kvmd crash if it is not connected or is disconnected during runtime.

**Describe the solution you'd like**

* Prevent kvmd from crashing if a device is not found or is disconnected during runtime.
* Introduce a flag to mark removable devices (removable: yes/no). Provide the ability to visualize this in the frontend, e.g. by using special input pin name like "status".

Suggested example configuration:

```
kvmd:
    gpio:
        drivers:
            atx_port:
                type: gpio
                device: /dev/removable-gpio-extension-device
                removable: yes
        scheme:
            gpio_status_led:
                driver: atx_port
                pin: status
                mode: input
```

**Describe alternatives you've considered**

Sorry, don't know of any alternatives.

**Additional context**

Got a bug report (https://github.com/zappanaut/pikvm-usb-atx-ctrl/issues/1) and thought it better fits here as a feature request.

mdevaev, 2023-02-25:
Hello. So, this device is working like pure gpio extender kinda `/dev/gpiochipX`?

zappanaut, 2023-02-26:
Hello! Yes, exactly. In this particular case the dln2 device driver is used. I am also using a symlink for the device to reference it via its serial number, but that should not make any difference.

mdevaev, 2023-02-26:
Please show what the exception that occurs during KVMD is running when you disconnect the USB.

zappanaut, 2023-02-26:
This is the error when disconnecting the device during runtime:
```
Exception in thread Thread-2 (__run):
Traceback (most recent call last):
  File "/usr/lib/python3.10/threading.py", line 1016, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.10/threading.py", line 953, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/lib/python3.10/site-packages/kvmd/aiogp.py", line 119, in __run
    for (pin, value) in zip(pins, lines.get_values()):
OSError: [Errno 19] No such device
```

and the error if the device is not connected on startup:
```
Traceback (most recent call last):
  File "/usr/bin/kvmd", line 9, in <module>
    main()
  File "/usr/lib/python3.10/site-packages/kvmd/apps/kvmd/__init__.py", line 75, in main
    KvmdServer(
  File "/usr/lib/python3.10/site-packages/kvmd/apps/kvmd/server.py", line 263, in run
    comp.sysprep()
  File "/usr/lib/python3.10/site-packages/kvmd/apps/kvmd/ugpio.py", line 283, in sysprep
    driver.prepare()
  File "/usr/lib/python3.10/site-packages/kvmd/plugins/ugpio/gpio.py", line 86, in prepare
    self.__chip = gpiod.Chip(self.__device_path)
FileNotFoundError: [Errno 2] No such file or directory
```

mdevaev, 2023-02-26:
Attach the full log please as file.

zappanaut, 2023-02-26:
Attached is the complete output log when starting kvmd on the command line.
[kvmd.log](https://github.com/pikvm/pikvm/files/10833817/kvmd.log)

mdevaev, 2023-02-26:
Got it, thank you.