pikvm/pikvm #947: Wrong mapping of @ char when pasting text with german keyboard layout

piezza, 2023-02-18:
Pasting text works in general but @ character in german keymap gets pasted as ". I think in english layout @ character is Shift+2. But in german layout Shift+2 is mapped to ". When pasting @ it should be mapped to AltGr+Q key in german layout.

mdevaev, 2023-02-26:
Hello. Did you try changine the keyboard layout in the paste menu? Just to be sure.

piezza, 2023-02-26:
I tried a bit more and it seems all characters which need to be accessed with AltGr have issues. I pasted following into a german windows:
[image]
resulting in this pasted text:
[image]
marked characters are the special ones which need AltGr on a german keyboard.

mdevaev, 2023-02-26:
I'll check, thank you.

mdevaev, 2023-03-23:
Apparently I fixed , but I have a question about the German keyboard layout. Is there some way to enter the `^` character? I tried the German layout on my Windows and didn't understand how to do it. I used the upper left key (to the left of `1`), and I cant enter only a two `^^` characters by double-clicking. How it should work?

piezza, 2023-03-23:
This is the crap German keyboard layout. Key `^` (the one left of `1`) and `´`  (the one left of backspace) are dead keys. German keyboards have this behavior that you need to press twice and then get the character two times. So I think this can't be fixed that easy.
What also may work is sending Alt+KeyCode. When I keep Alt pressed and enter 94 on the keypad I get also the `^` character (see https://www.alt-codes.net/).

mdevaev, 2023-03-23:
I see, thanks for the explanation. I've released the new version 3.208 with the fix, please try it.

mdevaev, 2023-03-23:
Was it working before update?

piezza, 2023-03-23:
Seems you are correct, also did not work before with the `€` sign.

mdevaev, 2023-03-23:
I found out the reason for the absence of the euro sign, I'll try to fix it.

mdevaev, 2023-03-23:
Okay, try 3.209. Euro should work now.

mdevaev, 2023-03-23:
Perfect. Then it's solved, I guess.