pikvm/pikvm #948: PiKVM Handbook   MSD   Wim.boot for over 2.2G

peacokswiss, 2023-02-20:
https://github.com/xe5700/kvmd-armbian/issues/12

read last post,   and expand your   Handbook for MSD and Windows  pls...

thx

arch1mede, 2023-02-25:
Hi, this is from a community member which we do not have any control over and that person does not monitor this github and at this time, no other SBC's are officially supported. I would recommend you open a request from their github and remove this.

peacokswiss, 2023-02-25:
I don't need you,  have you checked?

peacokswiss, 2023-02-25:
The best ist you remove your self ok
regards

mdevaev, 2023-02-25:
Hello. It was very difficult for me to understand the actual sequence of actions. As far as I can see, this is a special case. Thank you for the information, but we accept edits in the form of a ready-made pull request since we do not have the human resource to investigate, verify and describe each particular case.

peacokswiss, 2023-02-25:
then you make such remarkt,   i will not do...   
in respekt,  by

closed 2023-02-25 by mdevaev

peacokswiss, 2023-02-25:
i fick my self,,    
http://www.industrie-optimierer.ch/tmp/PiKVM/PiKVM_OrangePizero2/Readme.txt

peacokswiss, 2023-02-25:
some mistakes,,,  
but i make big ing stuff, in heavy Indsturie  ABB,  over 600Ampere per phase,   
so i dont need you chineese intro...
i know what i do..

peacokswiss, 2023-02-26:
If i talking about   one   1 TW    TerraWatt,    
is not importend your kidi sytem
not even a discusions...   only by alarm...status
don't talk again to me pls....  thx

peacokswiss, 2023-02-26:
Wennich was schreibe bleibt es,,,
ich finde auch eine unverstämte Frechheit mcih anzuficken   über email wegen git
so was doofes ist mir noch nie unter gekoommen,,,,
Ich code alles ,  alles von rockwell siemens, c bis böses Bash...   
halt bloss die Fresse..    wennich hässig werde.... dann ist das chinesen leben in Europa tot
mach mich ja nicht hässig
sicher

peacokswiss, 2023-02-26:
Soll ich noch ein Arschloch Papier und dir um den Arsch  wixxen.....
ist es das was du willst,  hier ist die Schweiz,  ich mach es....

peacokswiss, 2023-02-26:
Du Arschoch schon selbprogramierende  Betriebssyteme gehsehen,  ich schon in den 70er
Ich geb Dir ein Tipp,.  halt dich Fern von mir , wenn du länger leben willst,  ok  kappiert !!!