pikvm/pikvm #949: Clarification on how MSD works (trying to emulate empty USB to get data off server)

tan-wei-xin-alez, 2023-02-20:
**Describe the bug**
I've got a closed source Linux-based operating system running on a server that exports logs via USB so I can't install any software like ssh etc. or access it via command line since I don't have permission.

I was originally following the instructions at https://docs.pikvm.org/msd/#how-to-create-rw-flash-drive but was not successful.

I then went and used https://www.imgburn.com/ to create an .img file which was successfully detected by the server running the closed source OS but I could not write anything to it

Could I get clarification whether the above is possible without root access to the server being controlled by the PiKVM?
Because I was under the assumption that this should be completely possible (everything else seems to work fine).
Am I just doing it wrong by creating a writable-only .img file? I double checked the default settings on the imgburn application and it's "DVD-R/RW".
Or should I be using some other software on windows if the instructions at https://docs.pikvm.org/msd/#how-to-create-rw-flash-drive fail to work?

**To Reproduce**
1. Login to pikvm and go to `https://pikvm/extras/webterm/ttyd/`
2. Run `su -` in the web terminal to become root
3. Run `rw && kvmd-otgmsd -i 1` which throws the following error
```
Traceback (most recent call last):
  File "/usr/bin/kvmd-otgmsd", line 9, in <module>
    main()
  File "/usr/lib/python3.10/site-packages/kvmd/apps/otgmsd/__init__.py", line 104, in main
    print("Image file: ", (get_param("file") or "<none>"))
  File "/usr/lib/python3.10/site-packages/kvmd/apps/otgmsd/__init__.py", line 88, in <lambda>
    get_param = (lambda param: _get_param(config.otg.gadget, options.instance, param))
  File "/usr/lib/python3.10/site-packages/kvmd/apps/otgmsd/__init__.py", line 42, in _get_param
    with open(_get_param_path(gadget, instance, param)) as param_file:
FileNotFoundError: [Errno 2] No such file or directory: '/sys/kernel/config/usb_gadget/kvmd/functions/mass_storage.usb1/lun.0/file'
```
4. Resort to using https://www.imgburn.com/ to create .img file via the following process
  a. [image]
  b. [image]
  c. Select any random folder as source
  d. Write a destination path with .img extension
  e. [image]
5. Upload via the KVM webUI at https://pikvm/kvm/#
[image]
6. Name of volume within .img file shows up on server side but no writing is possible

**Expected behavior**
Should be able to emulate a USB and get data off the server (works perfectly fine with a normal USB flash drive).

**Desktop (please complete the following information):**
 - OS: `Get-ComputerInfo | select WindowsProductName, WindowsVersion, OsHardwareAbstractionLayer`
```
WindowsProductName WindowsVersion OsHardwareAbstractionLayer
------------------ -------------- --------------------------
Windows 10 Pro     2009           10.0.19041.2251
```
 - Browser: `(Get-Item 'C:\Program Files (x86)\Microsoft\Edge\Application\msedge.exe').VersionInfo`
```
ProductVersion   FileVersion      FileName
--------------   -----------      --------
108.0.1462.46    108.0.1462.46    C:\Program Files (x86)\Microsoft\Edge\Application\msedge.exe 
```

**PiKVM info:**
  - Raspberry Pi board version: `cat /proc/device-tree/model`
```
Raspberry Pi 4 Model B Rev 1.5
```
  - PiKVM platform: v3 HAT pre-assembled on Raspberry Pi 4 (2 GB) from https://www.elektor.com/pikvm-v3-kvm-over-ip-for-raspberry-pi-4-pre-assembled
  - KVMD version: `pacman -Q | grep kvmd`
```
kvmd 3.155-1
kvmd-fan 0.22-1
kvmd-oled 0.13-1
kvmd-platform-v3-hdmi-rpi4 3.155-1
kvmd-webterm 0.42-1
```
  - uStreamer version: `pacman -Q | grep ustreamer`
```
ustreamer 5.24-1
```
  - Linux kernel: `uname -a`
```
Linux pikvm 5.15.68-2-rpi-ARCH #1 SMP Fri Oct 7 06:33:56 MSK 2022 armv7l GNU/Linux
```

mdevaev, 2023-02-25:
Hello.

> kvmd-otgmsd -i 1

You should use `0` instead of `1` for the MSD.

But anyway :) As far as I understand, you performed further actions through the web interface. You did everything right except that the img file should have been formatted in FAT32, that is, you needed to create an image of a flash drive, not a DVD. Next, the emulation mode in the Drive menu in Web UI should be Flash. After you configure Flash, you should turn off and re-enable the main USB breaker in the System menu to perform physical reinitialization of the USB device. After that, your host will see a regular USB flash drive and you can simply copy the files there.

tan-wei-xin-alez, 2023-02-26:
@mdevaev thanks for the reply

> You should use 0 instead of 1 for the MSD.

Ah, then could you explain what
> Drive 0 represents a drive that is controlled via a web interface and API. Don't use it with kvmd-otgmsd if you don't know exactly what you're doing.
means at https://docs.pikvm.org/msd/#how-to-create-rw-flash-drive?
Should I follow all the steps there and use 0 instead of 1?

mdevaev, 2023-02-26:
I will correct the instructions, it refers to the old versions. In the new version, you can use 0 because the API now supports read-write storage.

tan-wei-xin-alez, 2023-02-26:
@mdevaev ok, I followed the instructions at https://docs.pikvm.org/msd/#how-to-create-rw-flash-drive but with 0 instead of 1 and now I get this

[image]

Should I be using `dd if=/dev/zero of=/root/flash.img bs=1M count=1000 status=progress` or some other command to create an empty .img file? Because I need to write to it, does the warning message not matter?

mdevaev, 2023-02-26:
Everything is as it should be, just follow the instructions. The warnings only tell you to be careful.

tan-wei-xin-alez, 2023-02-28:
Ok, I'll just leave this here for any future users that might bump into the same problem, maybe it can also be added to the docs later

<ul>
  <li><code>dd if=/dev/zero of=/root/flash.img bs=1M count=1000 status=progress</code> alone as shown at at <a href='https://docs.pikvm.org/msd/#how-to-create-rw-flash-drive'>How to create RW flash drive</a> is not enough for creating an .img file that can be mounted and read on a windows machine, the following is also needed to create a FAT32 partition within the file
<br><br>
<pre><code>dd if=/dev/zero of=&lt;path_to_img&gt; bs=1M count=1000 status=progress
(echo o; echo n; echo ""; echo ""; echo ""; echo t; echo c; echo w) | fdisk &lt;path_to_img&gt;
mkfs.vfat &lt;path_to_img&gt;</code></pre>
as seen at <a href='https://fejlesztek.hu/create-a-fat-file-system-image-on-linux/'>Create a FAT file system image on Linux</a>
<br>
(<code>fdisk</code> one-liner from <a href='https://unix.stackexchange.com/questions/480369/automatize-partition-creation-with-fdisk/480393#480393'>automatize partition creation with fdisk</a>)

Trying to mount the file on windows, however, will probably throw a "disk is corrupted" error so use something like <a href='https://www.osforensics.com/tools/mount-disk-images.html'>OSFMount</a> instead
  </li>
  <li>For the poor souls who need a windows solution, it takes a bit more messing around but it is possible
    <br><br>
    <ol type='1'>
      <li> Create a .vhd file using windows "Disk Management" tool as seen at <a href='https://www.windowscentral.com/how-create-and-set-vhdx-or-vhd-windows-10'>How to create and set up a virtual hard disk on Windows 10</a> (<b>REMEMBER</b> to allocate at least one partition within it)</li>
      <li> Convert the .vhd file into an .img file using <code>VBoxManage clonehd --format raw &lt;path_to_vhd&gt; &lt;path_to_img&gt;</code> as mentioned at <a href='https://superuser.com/questions/1079797/convert-vhd-to-img-on-windows/1079807#1079807'>Convert .vhd to .img on windows?</a></li>
      <li> Again, trying to mount the file on windows will probably throw a "disk is corrupted" error so use something like <a href='https://www.osforensics.com/tools/mount-disk-images.html'>OSFMount</a> instead</li>
    </ol>
  </li>
</ul>

closed 2023-02-28 by tan-wei-xin-alez