pikvm/pikvm #950: Render problems with MSI BIOS

etkaar, 2023-02-22:
Using a Raspberry Pi 4 with a CSI bridge (Geekworm) the BIOS of a modern B660 MSI mainboard looks a bit awkward.

```
Mainboard: MSI B660M MAG MORTAR (MS-7D42)
BIOS Version: E7D42IMS.A00
BIOS Build Date: 14th December 2021
Resolution: 1280x720
```

```
Linux kvm 5.15.68-3-rpi-ARCH #1 SMP Mon Oct 31 20:56:54 MSK 2022 armv7l GNU/Linux
```

[image]

arch1mede, 2023-02-25:
Hi, have you attempted any of the [edid](https://docs.pikvm.org/edid/) changes to see if they help?

mdevaev, 2023-03-02:
No answer, so I'm closing it. Please use discord for similar questions.

closed 2023-03-02 by mdevaev

etkaar, 2023-03-02:
I didn't forget it ☺️ I temporarily use the PiKVM on another machine with a different mainboard, so I haven't been able yet to test it. I will soon post my results regarding this issue.

etkaar, 2023-03-27:
Hi!

With following custom EDID (see [Tuning HDMI EDID - PiKVM Handbook](https://docs.pikvm.org/edid/#default-edid)) the MSI BIOS works:

```
1920x1080 as preferred. Useful for Gigabyte GA-H77-DS3H, Intel NUC or [ASRock H670 PG Riptide]
```
```
00FFFFFFFFFFFF005262888800888888
1C150103800000780AEE91A3544C9926
0F505425400001000100010001000100
010001010101D32C80A070381A403020
350040442100001E7E1D00A050001940
3020370080001000001E000000FC0050
492D4B564D20566964656F0A000000FD
00323D0F2E0F000000000000000001C4
02030400DE0D20A03058122030203400
F0B400000018E01500A0400016303020
3400000000000018B41400A050D01120
3020350080D810000018AB22A0A05084
1A3030203600B00E1100001800000000
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000045
```

However, no mouse support in BIOS, but in GRML. Is not really relevant in my case, but what could be the reason the mouse does not work?

arch1mede, 2023-06-11:
Hi, I have seen this same behavior even with commercial solutions, perhaps the different mouse modes will help. https://docs.pikvm.org/mouse