pikvm/pikvm #953: Rendering issue when in the Bios screen

ViciousWall, 2023-02-26:
**Describe the bug**
First of all thank you for the hard work that is put in to make this. I have installed PiKVM HDMI USB on my raspberry pi 4 and I am able to access the GUI and see my server terminal. The only issue is when I am in the Bios and select any option it just gets messy like you see in the screenshot and also shows a different language when my default language is English. Unfortunately I am unable to change the BIOS to CSM so I am unsure what I should be doing next. Any help on this would be appreciated. Thanks

**Expected behavior**
I expected it to show me the BIOS as I see when I connect a display to my server

**Screenshots**
See Screenshot

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Browser Firefox, Chrome and Brave

**PiKVM info:**
  - Raspberry Pi board version - RPI 4
  - PiKVM platform - v2 HDMI USB
  - Video capture type  - HDMI USB connection
  - KVMD version - kvmd 3.199-1
    kvmd-fan 0.25-1
    kvmd-oled 0.19-1
    kvmd-platform-v2-hdmiusb-rpi4 3.199-1
    kvmd-webterm 0.43-1

  - uStreamer version: ustreamer 5.37-1
  - Linux kernel: Linux pikvm 5.15.68-3-rpi-ARCH #1 SMP Mon Oct 31 20:56:54 MSK 2022 armv7l GNU/Linux

**Additional context**
Add any other context about the
[image]
 problem here.

mdevaev, 2023-02-26:
Hi. I'm sorry, but there's nothing we can do here. You're using HDMI USB, which does not have the ability to configure EDID (the monitor metadata). We could make some settings for CSI bridge, but here we literally have no control over this black box.

ViciousWall, 2023-02-26:
All good I will have a look at CSI bridge option then. Thanks for your response.

closed 2023-02-26 by ViciousWall