pikvm/pikvm #954: No such file or directory: '/sys/class/rtc/rtc0/since_epoch'

bit2pixel, 2023-03-01:
**Describe the bug**

I'm using a regular Raspberry Pi 4 with the PiKVM Hat and I'm seeing the following log: 

```
[2023-03-01 10:22:29 kvmd-watchdog.service] --- kvmd.apps.watchdog                INFO --- Running watchdog loop on RTC0 ...
[2023-03-01 10:22:29 kvmd-watchdog.service] --- Traceback (most recent call last):
[2023-03-01 10:22:29 kvmd-watchdog.service] ---   File "/usr/bin/kvmd-watchdog", line 9, in <module>
[2023-03-01 10:22:29 kvmd-watchdog.service] ---     main()
[2023-03-01 10:22:29 kvmd-watchdog.service] ---   File "/usr/lib/python3.10/site-packages/kvmd/apps/watchdog/__init__.py", line 121, in main
[2023-03-01 10:22:29 kvmd-watchdog.service] ---     options.cmd(config.watchdog)
[2023-03-01 10:22:29 kvmd-watchdog.service] ---   File "/usr/lib/python3.10/site-packages/kvmd/apps/watchdog/__init__.py", line 82, in _cmd_run
[2023-03-01 10:22:29 kvmd-watchdog.service] ---     _reset_alarm(config.rtc, config.timeout)
[2023-03-01 10:22:29 kvmd-watchdog.service] ---   File "/usr/lib/python3.10/site-packages/kvmd/apps/watchdog/__init__.py", line 58, in _reset_alarm
[2023-03-01 10:22:29 kvmd-watchdog.service] ---     now = _read_int(rtc, "since_epoch")
[2023-03-01 10:22:29 kvmd-watchdog.service] ---   File "/usr/lib/python3.10/site-packages/kvmd/apps/watchdog/__init__.py", line 47, in _read_int
[2023-03-01 10:22:29 kvmd-watchdog.service] ---     with open(_join_rtc(rtc, key)) as value_file:
[2023-03-01 10:22:29 kvmd-watchdog.service] --- FileNotFoundError: [Errno 2] No such file or directory: '/sys/class/rtc/rtc0/since_epoch'
```

**To Reproduce**
Steps to reproduce the behavior, like:
1. Go to 'https://[PiKVM-IP]/api/log?seek=3600&follow=1'
2. See error

**Expected behavior**
No error.

**Desktop (please complete the following information):**
 - OS: MacOS
 - Browser Brave

**PiKVM info:**
  - Raspberry Pi board version: RPi 4
  - PiKVM platform: Latest
  - Video capture type: HDMI
  - KVMD version:
  kvmd 3.200-1
  kvmd-fan 0.25-1
  kvmd-oled 0.19-1
  kvmd-platform-v3-hdmi-rpi4 3.200-1
  kvmd-webterm 0.43-1

  - uStreamer version: ustreamer 5.38-1
  - Linux kernel: Linux pikvm 5.15.68-3-rpi-ARCH #1 SMP Mon Oct 31 20:56:54 MSK 2022 armv7l GNU/Linux

mdevaev, 2023-03-01:
Hello. Could you take a photo of the HAT?

bit2pixel, 2023-03-01:
Hi! I won't be able to until tonight but here's the exact model I purchased:

Geekworm KVM-A3 Kit with Raspberry Pi 4 Case Compatible with PiKVM ATX HDMI CSI for Raspberry Pi 4 Model B Only(Not Include Raspberry Pi) 

https://www.amazon.com/dp/B0B5R37TT3

I have the latest updates and I installed NeoVim, just a heads up in case that's causing any conflicts.

mdevaev, 2023-03-01:
Sorry, but this is not a compatible device, and we don't support it. We support either DIY or V3+ devices. You should contact the manufacturer.

closed 2023-03-01 by mdevaev