pikvm/pikvm #956: Reproducible black screen issue

etkaar, 2023-03-02:
I have a machine with Windows 10 installed. Once I joined a Linux desktop (in my case *Zorin OS*) and then restart the PC, I get a black screen (not the "NO SIGNAL" message").

- A restart of PiKVM does not work.
- A restart of the PC does not work.
- Replugging the HDMI cable does not work.
- Resetting stream or HID does not work.

What instead works:

- Plugging out the HDMI cable, connect it to a monitor, restart the PC and then connect it back to the PiKVM.

I assume that this might be a EDID related issue.

Is there any function to reset this? In a live production environment I can't of course unplug the HDMI and connect it to a monitor.

mdevaev, 2023-03-02:
> Once I join a Linux desktop environment

How? Where? I don't understand.

Try this: https://docs.pikvm.org/edid

etkaar, 2023-03-02:
> > Once I join a Linux desktop environment
> 
> How? Where? I don't understand.
> 
> Try this: https://docs.pikvm.org/edid

What I mean: Flashing an ISO image of ZorinOS (which is a Linux distribution) to USB or using the emulated Mass Storage Drive and then starting it. That all works – but when I then restart the server, I get only a black screen.

etkaar, 2023-03-03:
Closing this for now since it is unclear if this might be a BIOS issue.

closed 2023-03-03 by etkaar