pikvm/pikvm #957: No way to disable iptables DROP in otgnet

tarasglek, 2023-03-05:
https://github.dev/pikvm/kvmd/blob/c98a18aa7e21dd2bfd3f76ebc2e0c056ef989759/kvmd/apps/otgnet/__init__.py#L114 unconditionally issues `-A INPUT -i usb0 -j DROP` and there is no way to undo it.

I would like to be able to use pikvm as a jumphost over otgnet to connect to computer without needing to enable routing.

mdevaev, 2023-03-07:
Hello. Did you try allow_tcp?
```yaml
otgnet:
    firewall:
        allow_tcp: [22]
```

mdevaev, 2023-03-13:
Sup?

mdevaev, 2023-03-21:
Sup?

tarasglek, 2023-03-24:
Sorry for slow response.
No change. This allows:
`-A INPUT -i usb0 -p tcp -m tcp --dport 22 -j ACCEPT` 
so i can ssh into pikvm from my host. Can't go from pikvm into my server.

mdevaev, 2023-03-25:
Please update KVMD to 3.210, should be fixed now

tarasglek, 2023-03-26:
works great, thanks!

closed 2023-03-26 by tarasglek

mdevaev, 2023-03-26:
Ur welcome