pikvm/pikvm #960: Please add the ability to disable HTTPS and access via HTTP

ipv6hack, 2023-03-09:
I want to create an environment where PIKVM is used from the Internet via IIS reverse proxy, so I want to be able to access PIKVM via HTTP instead of HTTPS!
I would like to implement a function to switch between HTTPS and HTTP for such an access method.
Please help me.

mdevaev, 2023-03-09:
Hello. Edit `/etc/kvmd/nginx/nginx.conf`, comment two `server` sections, and add the single one:
```nginx
	server {
		include /etc/kvmd/nginx/listen-http.conf;
		include /etc/kvmd/nginx/certbot.ctx-server.conf;
		include /etc/kvmd/nginx/kvmd.ctx-server.conf;
		include /usr/share/kvmd/extras/*/nginx.ctx-server.conf;
	}
```

closed 2023-03-09 by mdevaev

ipv6hack, 2023-03-09:
Worked well with HTTP
Thank you so much!
It helped me a lot!

mdevaev, 2023-03-10:
:ok_hand: