pikvm/pikvm #964: Orange Keyboard & Mouse Whenever KVM Is connected

jvanderaa, 2023-03-18:
**Describe the bug**
A clear and concise description of what the bug is.

Whenever I have a KVM in the middle following the set up defined https://blog.ktz.me/use-1-pikvm-instance-to-control-4-systems/, the controls are orange

**To Reproduce**
Steps to reproduce the behavior, like:
1. Go to KVM host
2. Click on the host, see the video
3. See orange keyboard and mouse

When I bypass the KVM everything is working fine for the single host.

**Expected behavior**

Expected control of the mouse

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Desktop (please complete the following information):**
 - OS: Mac
 - Browser Edge
 - Version Latest

**PiKVM info:**
  - Raspberry Pi board version [e.g. RPi 4] RPi 4
  - Video capture type [e.g. CSI bridge] CSI Bridge
  - KVMD version: `pacman -Q | grep kvmd`
  - uStreamer version: `pacman -Q | grep ustreamer`
  - Linux kernel: `uname -a`
```
[root@pikvm ~]# pacman -Q | grep kvmd
kvmd 3.205-1
kvmd-fan 0.25-1
kvmd-oled 0.22-1
kvmd-platform-v2-hdmi-rpi4 3.205-1
kvmd-webterm 0.43-1
[root@pikvm ~]# pacman -Q | grep ustreamer
ustreamer 5.38-1
[root@pikvm ~]# uname -a
Linux pikvm 5.15.68-3-rpi-ARCH #1 SMP Mon Oct 31 20:56:54 MSK 2022 armv7l GNU/Linux
```

**Additional context**
Add any other context about the problem here.
I've attempted multiple cables in every position. The only thing different hardware wise compared to the blog is the power splitter. It is from pishop instead of the affiliate linked item, which would take a month to get back.

Hoping for some guidance on how to t-shoot it a bit more. Will be glad to dig in, but I'm not quite sure where to go next from a logging perspective. For my initial strong desire for the solution, I only need a single host right now. So I can bypass, but have invested in the KVM and would like to see if I'm doing something just terribly wrong in there.

Thank you in advance!

arch1mede, 2023-03-18:
Hi, this might be a bit easlier if you could join the [Discord](https://discord.gg/bpmXfz5) so I could assit you faster. 

In the mean time, as a first step, have you followed any of the example wiring pics as outlined in the [Ezcoo doc](https://docs.pikvm.org/ezcoo/)?

If those icons are not green, its likely a cable issue and you will need to test each cable to be sure they are not power only. I'd hate to think this is a defective ezcoo however, there is always that possibility.

jvanderaa, 2023-03-18:
Discord is the route to go!

closed 2023-03-18 by jvanderaa