pikvm/pikvm #966: Hiding mouse lag with a more visible local mouse pointer

gourdo1, 2023-03-22:
Probably the most annoying aspect of relying on a pikvm to navigate my Windows laptop is that the remote mouse lags quite a bit.

However, I've noticed that if the local mouse cursor was visible while the remote cursor hidden, it would actually completely resolve the perception of lag -- without having to actually address the underlying causes (i.e. processing and network latency, etc.)

In that vane, I've switched to an invisible mouse cursor on my remote device (a simple 4kb .cur file download for Windows) which effectively hides the lag that the remote mouse cursor experiences. The next step to complete the illusion would be to replace the current translucent disk that pikvm displays locally (which is often times difficult to see and track) with something closer to a Windows mouse cursor that's a lot more visible.

To this end, I'm wondering whether a mouse cursor style toggle is something you'd support in the pikvm settings menu.

mdevaev, 2023-03-23:
Removing the blue dot will only make things worse. The delay will not go anywhere, but positioning the cursor will turn into a very unpleasant task, because you will constantly miss your target. Moreover, it is enough to point the blue dot at the element and not wait for the real cursor to catch up with it. If you click the mouse button when the blue dot is where you need it, the coordinates will be transmitted instantly before the click.

I recommend just adapting to this mechanics rather than fighting it. It will be worse without it, tested by many users.

gourdo1, 2023-03-23:
@mdevaev To be clear, I'm not asking for a way to remove the blue dot. Quite the opposite. I'm asking for an alternative cursor that's more visible than the blue dot. Something like the attached...

I would manually suppress the remote cursor myself.
[image]

mdevaev, 2023-03-23:
Okay, I didn't understand you correctly first time.

You can change UI elements by user CSS. Edit `/etc/kvmd/web.css` and add something like this:
```css
div.stream-box-mouse-enabled {
    cursor: unset;
}
```

mdevaev, 2023-03-24:
Solved I guess.

closed 2023-03-24 by mdevaev