pikvm/pikvm #967: How to simulate a disk or other form under Windows system so that Windows can upload and download files?

TelDragon, 2023-03-24:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

How to simulate a disk or other form under Windows system so that Windows can upload and download files?
如何在windows系统下模拟磁盘或其它形式，使其windows可以进行上传下载文件？

This is not a simulation of the type of file drive letter that a CD or DVD wants.
在这里不是模拟cd 也不是模拟DVD 想要磁盘类型的文件盘符那种。

etkaar, 2023-03-25:
Hi! Could you please describe your issue in English? :)

mdevaev, 2023-03-26:
You need to create an empty image with fat filesystem and connect it with rw mode in msd ui.

https://docs.pikvm.org/msd/

closed 2023-03-26 by mdevaev