pikvm/pikvm #980: kvmd-certbot does not seem to copy authentication files to the temporary working directory

vipergts450, 2023-03-26:
**Describe the bug**
I use the certbot-dns-google plugin to manage SSL certificates. Initially, the certificate pulls fine, but the subsequent automatic renewals via certbot-timer.service always seem to fail due to an inability to find the auth file I created. The problem seems to stem from the process via which kvmd-certbot sets up the temporary working directory, namely `cp -a "$cur"/{config,work,logs} "$tmp"` on line 111. This seems to not copy the auth file at the top of the `/var/lib/kvmd/pst/data/certbot/runroot` directory.

**To Reproduce**
1. After initializing kvmd-certbot via the documentation attempt an automatic or manual renewal when within the expiry timeframe.
2. Run `kvmd-certbot renew`
```
[root@pikvm ~]# kvmd-certbot renew
Saving debug log to /tmp/kvmd-certbot/runroot/logs/letsencrypt.log

- - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - -
Processing /tmp/kvmd-certbot/runroot/config/renewal/<domain>.conf
- - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - -
Renewing an existing certificate for <domain>
Failed to renew certificate <domain> with error: File not found: /tmp/kvmd-certbot/runroot/<auth-file>

- - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - -
All renewals failed. The following certificates could not be renewed:
  /tmp/kvmd-certbot/runroot/config/live/<domain>/fullchain.pem (failure)
- - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - -
1 renew failure(s), 0 parse failure(s)
Ask for help or search for solutions at https://community.letsencrypt.org. See the logfile /tmp/kvmd-certbot/runroot/logs/letsencrypt.log or re-run Certbot with -v for more details.
```
4. Certificate is not updated

**Expected behavior**
The auth file should also be copied from `/var/lib/kvmd/pst/data/certbot/runroot/` to the temporary directory, or the documentation should be updated to show which subfolder the authentication credentials should be in. Should I move my auth file to the directory `/var/lib/kvmd/pst/data/certbot/runroot/config/renewal/` subdirectory or some other location?

**PiKVM info:**
  - RPi 4B 8GB
  - PiKVM platform `v3-hdmi`
  - KVMD version: `kvmd 3.210-1`
  - uStreamer version: `ustreamer 5.38-1`
  - Linux kernel: `Linux pikvm 5.15.68-3-rpi-ARCH #1 SMP Mon Oct 31 20:56:54 MSK 2022 armv7l GNU/Linux`

mdevaev, 2023-03-29:
Could you make a patch for this? I don't have Google DNS to test.

vipergts450, 2023-03-29:
I feel like this issue may not be specific to Google DNS, but possibly also any provider which is set up with an authentication file stored in the top of `runroot`. For example, the PiKVM Wiki example for Cloudflare DNS says:
`2. Prepare the environment for the DNS plugin (place the auth data):

# kvmd-pstrun -- mkdir -p /var/lib/kvmd/pst/data/certbot/runroot
# kvmd-pstrun -- nano /var/lib/kvmd/pst/data/certbot/runroot/.cloudflare.auth
# kvmd-pstrun -- chmod 600 /var/lib/kvmd/pst/data/certbot/runroot/.cloudflare.auth`

If my issue is related to the `cp -a` line on line 111, then Cloudflare renewals may also get caught by the same problem due to where the documentation says to store the .auth file.

Has anyone else run into an automatic renewal issue due to this file placement?

Interestingly, running `kvmd-certbot -- -v renew` seems to get around this by not doing the copy and temp directory set up and running the renewal in the actual `runroot` folder.

I am happy to play around and try a patch, but also am surprised no one else has hit a renewal issue for similar set ups.

vipergts450, 2023-03-29:
Gave it a shot... I felt really silly that it was a one character diff, so I felt like I had to add a lot of descriptive language around it. 
[PR #125](https://github.com/pikvm/kvmd/pull/125).

mdevaev, 2023-03-29:
It's hard to say why it wasn't noticed, perhaps other plugins behave differently. Thanks for the patch :ok_hand:

closed 2023-03-29 by mdevaev