pikvm/pikvm #982: PiKVM won't boot after full system update

rmbleeker, 2023-03-30:
**Describe the bug**
PiKVM won't boot after a full system update. This happened to me twice now, first after updating my existing PiKVM installation during a regular round of updates. Because the system was then useless I downloaded and wrote a new image to the USB drive I use to run the OS from. I did a full system update after the re-image with the same result.

**To Reproduce**
Mount filesystem read-write with the `rw` command, perform a `pacman -Syu` and then `reboot`.
While booting you'll be greeted with the following error message:
```
:: mounting 'LABEL=PIROOT on read root
mount: /new_root: can't find LABEL=PIROOT.
You are now being dropped into an emergency shell.
```

**Expected behavior**
I would expect the Raspberry Pi to just start after performing the latest updates.

**PiKVM info:**
  - RPi 4
  - PiKVM v3
  - Linux kernel 5.15.68-3-rpi-ARCH

rmbleeker, 2023-03-30:
I used RPi Imager indeed and used v3-hdmi-rpi4-box-latest.img.xz as the image when I did the reinstallation today. However these things shouldn't be relevant since the original installation was about half a year old, updated a few times already and always working perfectly. It only broke after performing updates today, forcing me to reapply the image, with which I can reproduce the issue every time.

mdevaev, 2023-03-31:
May I ask you to flash it, perform full upgrade and attach the full pacman log? For some reason we couldn't reproduce the issue, it is weird :/

rmbleeker, 2023-03-31:
That's odd since I was able to reproduce the issue without fail every time after performing a full package upgrade after flashing a new image. To my knowledge I don't have any image customization options active in the RPi Imager, but I will check when I get home and also see if I can provide additional information, including a pacman log. I'm not too familiar with Arch so can you tell me where to look for the log if it's not in /var/log?

To complicate matters, in my impatience it looks like I damaged the mini-HDMI out on my Pi yesterday, so I can't hook up a monitor anymore and troubleshoot the issue from the console.

mdevaev, 2023-03-31:
I need not just a pacman log, I need a full console log. Just keep it and copy-paste.

rmbleeker, 2023-04-01:
Okay so I re-downloaded the image and this time used `dd` to flash it. Then I performed a full system update. Here's the pacman.log that I saved before rebooting and the console output. The result is the same, after the reboot command the system shuts down but doesn't boot.

[pikvm-console.txt](https://github.com/pikvm/pikvm/files/11130753/pikvm-console.txt)
[pikvm-pacman.log](https://github.com/pikvm/pikvm/files/11130754/pikvm-pacman.log)

mdevaev, 2023-04-01:
I don't understand what's wrong here :/ Could you try a different SD card?

rmbleeker, 2023-04-02:
I just tried a different SD card, an actual SD card in fact since I was using a USB thumb drive. Now it works without issues. This is strange to me since the USB drive was bought brand new for this purpose so it's only half a year old and shouldn't be defective already. But obviously there's nothing you can do about that. 

I'm truly sorry for wasting your time, I should have tried a different drive right away.

mdevaev, 2023-04-02:
I didn't pay attention to it right away either. It's good that this has been resolved.

rmbleeker, 2023-04-02:
Depends on your definition of resolved ;-)
Overnight I ran a thorough check of the USB drive that was causing the issue to find anything wrong with it, but there just isn't. So it's still a mystery to me why it's happening with the USB drive but not the SD card. I do think that this is more a general Arch Linux issue than a specific PiKVM issue. Do you think it would be useful to make the Arch Linux community aware of this issue and if yes, how should I go about doing that?

mdevaev, 2023-04-02:
I'll take care of it myself when I find out what the reason is.

rmbleeker, 2023-04-02:
I'll experiment some more and see if I can find out any more useful information.

mdevaev, 2023-04-02:
:ok_hand:

k-ronny, 2023-04-02:
Hello
I found out that the following change to the file "/etc/mkinitcpio.conf" solves this problem for me:

in `/etc/mkinitcpio.conf` change the line
```
HOOKS=(base udev block filesystems)
```
to
```
HOOKS=(base udev block filesystems fsck)
```

mdevaev, 2023-04-02:
This is strange, why...

k-ronny, 2023-04-02:
I don't know, but in the file `/etc/mkinitcpio.conf` the `fsck` hook is almost always used and in the [ArchWiki](https://wiki.archlinux.org/title/Mkinitcpio#Common_hooks) you can find this statement: _Usage of this hook is strongly recommended, and it is required with a separate /usr partition._

k-ronny, 2023-04-02:
Oh, I just read this in the [ArchWiki](https://wiki.archlinux.org/title/Fsck#Boot_time_checking), maybe it explains that?

mdevaev, 2023-04-02:
But this does not explain why labelled fs are not detected. I need to study the sources.

mdevaev, 2023-04-04:
Thanks. I'll study it.

rmbleeker, 2023-04-05:
> Just found out: If you use
> 
> ```
> root=/dev/disk/by-label/PIROOT
> ```
> 
> instead of
> 
> ```
> root=LABEL=PIROOT
> ```
> 
> it works again, also without the addition of `fsck` in the initramfs.

That's all well and good but the real mystery then becomes why the "LABEL" method keeps working when booting the Pi from an SD card but not when booting from a USB thumb drive.

rmbleeker, 2023-04-05:
Might it have something to do with the fact that I'm using a USB3 thumb drive on the USB 3.0 ports, which on the Raspberry Pi 4 work through a separate controller I believe? Maybe the kernel doesn't initialize the USB 3.0 controller quickly enough so that the root partition can not be found when it's needed?

sw25481, 2023-04-30:
Just hit exactly the same problem.  Will try the same work around and report back

I boot from a Samsung T5 Portable USB SSD on an RPI4.  This drive has always worked well with any distro and PiKVM for several months until the most recent pacman -Syu

Boot fails with mount: /new_root: can't find LABEL=PIROOT however you can mount the partition fine from the emergency screen by issuing "mount -L PIROOT /new_root/"

sw25481, 2023-04-30:
root=/dev/disk/by-label/PIROOT

worked for me also.  Thanks for the tip

mdevaev, 2023-05-21:
Besides enabling fsck, you should add `fsck.repair=yes` (maybe also `fsck.mode=force`) to `/boot/cmdline.txt`, otherwise, fsck will get stuck when critical problems are detected in interactive mode. Not sure if this completely disables interactive mode or not. Note: it is required to send the last number in fstab.

Another problem is that fsck corrects the time of the last superblock mount each time due to the kernel applying the time from the RTC after fsck. I don't know if this has any negative consequences, but we need to find a way to poll the clock before fsck.

Rather, I would prefer to enable fsk, but the question of interactivity, mount time and the existence of any side effects requires an answer.

* https://wiki.archlinux.org/title/Fsck#Boot_time_checking
* https://wiki.archlinux.org/title/mkinitcpio

ironicbadger, 2023-05-22:
Full system update borked my install. Fresh download of image and reflash brought me here. Still an issue.

mdevaev, 2023-05-22:
@ironicbadger do you use usb stick drive?

ironicbadger, 2023-05-22:
Yes usb boot

mdevaev, 2023-05-22:
@ironicbadger could you set `root=LABEL=PIROOT` and add another option `rootdelay=5`? Will it work without fsck hook and /dev/disk/by-label?

mdevaev, 2023-05-23:
Okay, I figured out what the problem is. USB does not have time to initialize before the scripts in initrd try to mount it. I will make an elegant workaround for this using a custom hook for mkinitcpio.

mdevaev, 2023-05-24:
Here the proper fix:
* Install the package `mkinitcpio-rootdelay`
* Set `LABEL=PIROOT` back.
* Run `mkinitcpio -p linux-rpi`.

mdevaev, 2023-05-25:
New images in https://files.pikvm.org have this feature enabled by default.

closed 2023-05-25 by mdevaev