pikvm/pikvm #984: How to upload local files to and download files from a mounted USB drive to the local location.

xinshaoyang, 2023-04-04:
I am currently studying IPKVM, which is a great project with powerful features.

I have a problem that I don't know how to solve, which is that I have mounted a USB drive but I don't know how to upload local files to it or download files from the USB drive to the local location. Can someone guide me? 

thank you

xinshaoyang, 2023-04-11:
Think you

mdevaev, 2023-04-11:
Hello. I'm sorry, I thought I was answering your question. Is it still relevant?

TelDragon, 2023-04-12:
First of all, thank you to the author for providing powerful ipkvm project functionality!

I also have the same needs or issues here.

Create an empty IMG image file as shown in the document and select writable for Windows mounting. This step has been implemented.

But my problem is that when it is possible to format (FAT) the mounted IMG simulation USB drive on the controlled end (Windows), the content can be copied in, and files can also be copied to Windwos.

But how can I read the content inside this IMG in the pikVM OS system?

xinshaoyang, 2023-04-12:
@mdevaev Hello, I haven't found a solution to it. Can you provide any relevant suggestions? thank you.

mdevaev, 2023-04-12:
You should use losetup and mount your image on PiKVM side.

xinshaoyang, 2023-04-22:
I tried using the following command, but it didn't work.
First,I use commands to mount to the client machine.
     kvmd-otgmsd -i 1 --set-rw=1 --set-cdrom=0 --set-image=/root/flash.img
after that,I use losetup synchronization parameters ,But I didn't get the result I wanted.The modified files on the client cannot be displayed in real-time in the/opt/tftpboot folder, and similarly, the modified files in the/opt/tftpboot folder cannot be displayed in real-time in the client.
     losetup /dev/loop0 /root/flash.img
     mount -o sync /dev/loop0 /opt/tftpboot

Is there a problem somewhere?   Thank you

mdevaev, 2023-04-22:
File systems are not designed to be used simultaneously in two OS. You must unmount it from one of the hosts, otherwise there will be content corruption.

closed 2023-08-10 by mdevaev