pikvm/pikvm #985: Documentation about video out (RPi4 board)

chri2, 2023-04-04:
I tried to find information about how the video-out is supposed to work and didn't find it. Maybe I've not looked in the right place, but I'd expected that it would be described somewhere [here](https://docs.pikvm.org/v3/#basic-setup) where there are the pictures of the board in **3. Connect PiKVM**.

On my PiKVM with the metal case the port is on the side and labled **"VIDEO OUT"**. I'd expect that the video of the device connected to "VIDEO IN" would be available at the **"VIDEO OUT"** to connect a local console. But there doesn't show anything when I connect a monitor.

Is this supposed to work like I think? Wouldn't it be nice to add the answer to the documentation?

And BTW - I would have loved to ask this in a support forum or chat, but see https://github.com/pikvm/pikvm/issues/792#issuecomment-1495423207 …

mdevaev, 2023-04-14:
Sup. The video output should have its own PiKVM console. Do you see it?

arch1mede, 2023-04-17:
Hi, just adding my 10 cents to this (inflation), video out will sometimes not show up till its restarted several times, unknown why. This can also be seen from other distro's so it actually might not be an arch linux thing. With that said, video out is from the rpi4 board so this will not be documented. Connecting the PiKVM is a separate activity.

**_I'd expect that the video of the device connected to "VIDEO IN" would be available at the "VIDEO OUT" to connect a local console_**
I am afraid there is no such thing. Think of the PiKVM as a physical monitor kb/mouse that can be accessed remotely. Video IN goes IN but only out through the webgui. Video OUT is from the console used for troubleshooting and is hardly ever in use, unless you like to see the console. Its quite boring really.

**_And BTW - I would have loved to ask this in a support forum or chat, but see https://github.com/pikvm/pikvm/issues/792#issuecomment-1495423207_**
There are no plans at this time to support another chat medium.

chri2, 2023-04-18:
> Sup. The video output should have its own PiKVM console. Do you see it?

After connecting a different monitor I got an output - a console as expected by your answer.

There seems to be problems with the signal quality of the rpi4 - it shows on one monitor and not on another (that works for other devices - including other sbc like rock64(pro) or cubox - well).
 
It would be nice if the expected output and the use case for the RPi hdmi out would be documented. If the RPi4 output issue is a known issue (didn't look for it, yet) it could be referenced there, too.

> Think of the PiKVM as a physical monitor kb/mouse

Thanks for the explanation! I just didn't *know* it before and had been wondering what kind of video-out I should expect.

arch1mede, 2023-04-30:
I will leave the final decision to the author but the video OUT is already well documented in other places, like Google, RPi Foundation etc...I cannot make comment on the behavior of other SBC's as we do not support other SBC's. 

There is only 1 use case for the video out and that's for console debugging as per any server OS. While convenient, its not necessary.

> If the RPi4 output issue is a known issue (didn't look for it, yet) it could be referenced there, too.

Its not a known issue, just something I have personally come across and how to overcomes this behavior.

flpydsk, 2023-06-15:
@chri2, Part of your issue may also be your HDMI Cable/setup. its quite possible you are running in to an edge case.

spoelstraethan, 2023-08-05:
Typically this is an issue because the Pi doesn't try to rescan and adjust the HDMI output ports unless you are running a desktop environment, so if you hook up a monitor that only supports 1080p or 1440p or 4k and the Pi was just outputting 480p or whatever the monitor won't display. I've found that 1080p monitors work extremely well and often can handle a little lower resolutions.

The best workaround is to plug in the monitor and trigger a reboot of the Pi, when it boots up it will probe the device's EDID and try to output a resolution that it says it can support.

mdevaev, 2023-08-09:
Added some note.

closed 2023-08-09 by mdevaev