pikvm/pikvm #986: PiKVM does not boot with device connected

chri2, 2023-04-06:
**Describe the bug**
When I switch on the power to my PiKVM the power led inside the case switches on and nothing happens - no output on the oled, no blinking of the `ACT` led.

**To Reproduce**
Steps to reproduce the behavior, like:
1. Connect your Librem5 thorugh a Baseus 8-in-1 usb hub to PD usb-c power, the usb-console of the PiKVM to the hub and an hdmi cable between PiKVMs `VIDEO IN` and the hubs hdmi output.
2. Connect the power to the PiKVM
3. wait
4. see error: power led is lit, but PiKVM doesn't start
5. disconnect the hdmi between PiKVM and usb hub
6. PiKVMs `ACT` led starts blinking, oled switches on, PiKVM boots
7. re-connect hdmi: everything works as expected

**Expected behavior**
PiKVM should boot while connected via hdmi to the usb-c hub (connected to the Librem5).

**PiKVM info:**
  - Raspberry Pi 4 Model B Rev 1.4
  - PiKVM v3 HAT (from Crowdfunding Campaign)
  - Video capture as in default configuration
  - KVMD version: kvmd 3.211-1
  - uStreamer version: ustreamer 5.38-1
  - Linux kernel: Linux pikvm 5.15.68-3-rpi-ARCH #1 SMP Mon Oct 31 20:56:54 MSK 2022 armv7l GNU/Linux

mdevaev, 2023-04-06:
Hello. This is a known issue. See (9) how to fix it: https://docs.pikvm.org/v3/#basic-setup

[image]

chri2, 2023-04-07:
Thanks, yes, that solved the issue.

I already had been through the basic-setup section using a different system. Changing the connected system seems to have triggered the bug. I'm not an engineer, but I'd like to understand why there is that jumper. 

I suppose that there is some downside to removing it - otherwise there'd be no reason to have it closed by default or have that jumper on the board at all.

Is there somewhere a description which reasons exist to have the jumper on the board and to have it closed by default?

BTW: I looked through the `Video` section of the documentation, because I had the vague recollection of having read something about booting and hdmi before. But I didn't go through the basic setup again to look for it.

mdevaev, 2023-04-07:
This jumper is there for paranoid reasons. Practice has shown that it does not get worse from its removal. Here is a full description of the problem: https://www.kickstarter.com/projects/mdevaev/pikvm-v3-hat/posts/3296343

chri2, 2023-04-07:
That's a great article! That information could be linked into the documentation.

Did you ever hear that opening the jumper led to any problems?

mdevaev, 2023-04-07:
Nop

closed 2023-04-08 by chri2

mdevaev, 2023-04-08:
I've also added KS link with the story to the doc.