pikvm/pikvm #989: Keyboard and Mouse unresponsive after host reboot, then display also doesn't work after trying to start new KVM

TokenRing, 2023-04-14:
**Describe the bug**
Hello, I used the pacman -Syu command a couple days ago to update my PiKVM with whatever the default repos offered to me. First time updating, only got the PiKVM a few days before that. Now, when I use the VNC session to initiate a reboot of the server, and then come back into Windows, I notice the display works, but the keyboard and mouse do not. If I then quit the VNC session and then try to go back into it, then neither the display, nor the keyboard/mouse work at all.

The display window title reads No Signal, 0 FPS on both MJPEG and H264 mode, and the keyboard and mouse icons in the KVM session top bar are yellow. The keyboard reads "keyboard captured, inactive/busy" and the mouse reads "mouse free, inactive/busy". The colors do not change when I move the mouse over the VNC display window; but the cursor changes to the hourglass (well, spinner - you know what I mean. Windows stopped using hourglass a long time ago).

CLI session works fine.

**To Reproduce**
Steps to reproduce the behavior, like:
1. Have PiKVM VNC session open
2. Reboot the Windows host
3. No mouse or keyboard, but displays.
4. Close VNC session.
5. Reopen.
6. Now, no display or KB/M.

**Expected behavior**
Keyboard, mouse, and display should work.

**Screenshots**

[image]

**Desktop (please complete the following information):**
 - OS: Accessing the https://pikvm session, Windows 10. Host OS PiKVM attached to, Windows Server 2016
 - Browser - accessing from both Firefox and Chrome has same result.
 - Version latest
 - VNC client (if used) N/A

**PiKVM info:**
  - Raspberry Pi board version [e.g. RPi 4] I purchased my PiKVM preassembled from PiShop.us, I assume the RPi4 with 2GB RAM?
  - PiKVM platform [e.g. v2-hdmi] I purchased my PiKVM preassembled from PiShop.us in April 2023, I don't know what this means. Current? Newest?
  - Video capture type [e.g. CSI bridge] I purchased my PiKVM preassembled from PiShop.us, I don't know what this means.
  - KVMD version: kvmd 3.211-1
kvmd-fan 0.25-1
kvmd-oled 0.22-1
kvmd-platform-v3-hdmi-rpi4 3.211-1
kvmd-webterm 0.43-1
  - uStreamer version: ustreamer 5.38-1
  - Linux kernel: Linux pikvm.local 5.15.68-3-rpi-ARCH #1 SMP Mon Oct 31 20:56:54 MSK 2022 armv7l GNU/Linux

**Additional context**
Seems similar to #315 but I have made no changes to the yaml override file mentioned.

TokenRing, 2023-04-14:
If I restart the PiKVM server (shutdown -r), then things work fine again - until the next time I need to reboot the host the PiKVM is connected to. Then the whole things starts again.

mdevaev, 2023-04-15:
Please try incognito session in the browser

TokenRing, 2023-04-16:
> .

Hi Maxim. Thanks for building such an interesting and useful product. As requested, tried Private Mode in Firefox, and Icognito mode in Chrome. No change; same results - "no signal" in video window, and mouse/keyboard icons in toolbar are yellow instead of green.

arch1mede, 2023-04-17:
Hi, I would like to ask if you could join us in [Discord](https://discord.gg/bpmXfz5), this kind of troubleshooting might take some time to figure out. This would also enable the community to assist if I am not able to respond right away.

I have personally never run into this issue where I have needed to restart/reboot the pikvm however I am not running 2016.

In the mean time, when this happens again, please run `journalctl -fu kvmd` and paste this into pastebin, then post the link so we can review it.

arch1mede, 2023-04-17:
On a side not, if you have the system sleep or set the display to sleep this also might pose an issue, have you [added](https://docs.pikvm.org/faq/?h=otg#usb-problems-keyboard-mouse-mass-storage-etc) this to your override.yaml file?

```
otg:
    remote_wakeup: true
```

TokenRing, 2023-04-17:
> In the mean time, when this happens again, please run `journalctl -fu kvmd` and paste this into pastebin, then post the link so we can review it.

Sure, here it is - https://pastebin.com/eq2ZsTy3

"persistent device timeout" - interesting?

Regarding "remote_wakeup: true" - this is a server machine that never sleeps at all. However, the display does sleep after I believe, the default 15 minutes of inactivity. But - the problem exhibits itself _immediately_ after the windows machine restarts, literally as I am watching the VNC session. Do you think this is relevant to me? At this time, I have made zero modifications to the override.yaml file.

I am no expert, but it does seem suspect that the mouse/keyboard icons are showing orange "inactive/busy" rather than green like they do when everything is working. I am wondering if the display not working is just a side effect of that.

arch1mede, 2023-04-18:
I honestly do not know what is causing it, its looking to me like this is a Target system issue. 

Whats the OS of the target? 
What is the make/model of the MB/server/desk that is being controlled?
Does this happen on a different desktop running plain windows?

Have you added that remote_wakeup statement to see if anything changes?

arch1mede, 2023-04-30:
Hi, are you still experiencing this same issue? If so I would ask that you join Discord so we can help troubleshoot.

TokenRing, 2023-05-02:
I'm terribly sorry I've just been busy with family stuff and haven't had time for fun tinkering/troubleshooting stuff. When I have time to pull the pikvm off the server its on I'll try it on my other PC to see if the problem is the same. I apologize for cluttering the board here but I just wanted to see if it was a known issue first of all (apparently it is not an easy fix!).

mdevaev, 2023-10-31:
This question has been hanging open for several months, so I'm closing it. If you want to continue the investigation, please use our [Discord](https://discord.gg/bpmXfz5) to contact support.

closed 2023-10-31 by mdevaev