pikvm/pikvm #992: Paste behaviour of '@' symbol different to keyboard input to client - UK

intellirat, 2023-04-17:
**Describe the bug**
While the '@' symbol is being correctly input to the client (w10 UK/gb input language) from the PiKVM console (with en/gb keymap), that same symbol is ignored in any pasted text.

**To Reproduce**
1. Typing the '@' symbol (as shifted ' ) works correctly - keycaps/map is correctly translated from PiKVM to client
2. Inputting a string in to the text/paste window show the correct input (e.g. 'email@address.com')
3. Pushing paste then replays the pasted text to the client, minus the '@' symbol (e.g. 'emailaddress.com')
4. 'ask paste confirmation' / 'hide input text' options have no effect. No other symbols seem to be affected.

**Expected behavior**
The '@' symbol should paste, as any other character does, assuming keymaps, and client language map. Even if they didn't match, I'd expect the incorrect symbol to be pasted, rather than nothing.

**Desktop (please complete the following information):**
 - OS: Host Windows 11, client Windows 10
 - Browser: Brave / Edge / Firefox all tried
 - Version: 1.50.114 / 112.0.1722.48 / 112.0
 - VNC client: n/a 

**PiKVM info:**
  - Raspberry Pi board version: Pi 4 4GB
  - PiKVM platform: v2-DIY
  - Video capture type: Geekworm Raspberry Pi HDMI IN Module, HDMI to CSI-2 CSI
  - KVMD version: 3.212-1
  - uStreamer version: 5.38-1
  - Linux kernel: 5.15.68-3-rpi-ARCH

**Additional context**
This has appeared in the last month or so - it was working correctly before this. Sorry, I wish I could be more precise.

mdevaev, 2023-04-27:
Hello. Did you try to change the target layout?
[image]

intellirat, 2023-04-27:
Hi, thanks for looking. I did change the host keymap, as described, and changed it back. 
For example, en-us will paste my UK string example@mail.com as example"mail.com (as you'd expect, but showing the @ character is present in the string). Changing back to en-gb pastes the string as examplemail.com again.

As far as I can see, the keymap looks standard (/usr/share/kvmd/keymaps/en-gb), with the relevant key:
# evdev 40 (0x28), QKeyCode "apostrophe", number 0x28
apostrophe 0x28
at 0x28 shift
dead_circumflex 0x28 altgr
dead_caron 0x28 shift altgr

mdevaev, 2023-04-27:
Fixed in KVMD 3.216

closed 2023-04-27 by mdevaev