pikvm/pikvm #996: Regular check of Discord server invitation - Invitation has expired

smalldragoon, 2023-04-18:
**Describe the bug**
The Invitation link for the Support discord channel has expired

**To Reproduce**
Steps to reproduce the behavior, like:
1. Go to 'https://pikvm.org/support/'
2. Click on 'click here to join the chat' 
3. Discord Window opens
4. See error : Invitation Link has expired

**Expected behavior**
Join Discord Server

**Desktop (please complete the following information):**
N/A, same behavior Desktop / Mobile

**PiKVM info:**
N/A

**Additional context**
Checked with 2 different Discord accounts

smalldragoon, 2023-04-19:
HI, Yes still not working here 
Here are some screenshots ( Desktop & Mobile ) . 
Desktop : 
[image]
Mobile : ( sorry in French )
[image]

mdevaev, 2023-04-19:
Try this: https://discord.gg/NG74RqV7FP

If it will not work, try this: https://stealthygaming.com/why-discord-unable-to-accept-invite/

smalldragoon, 2023-04-20:
Same result with the link 
I reviewed the 7 reasons, none seems matching on my side 
I re-tried with 2 accounts
[image]

and  2nd one : 
[image]

The error message on mobile indicate that " link is invalid or has expired "

smalldragoon, 2023-04-21:
I added a phone number , no changes . Are you sure I’m not banned ?

mdevaev, 2023-04-22:
I've just checked, you don't banned :/

smalldragoon, 2023-04-22:
I created a Discord Ticket.I read in some forum an issue with servers you joined, left and trying to rejoin.
One workaround is to join another server and try again. I managed to join with the 2nd account I created, it will work for me for now. Will keep you posted with result from Discord

mdevaev, 2023-04-22:
Okay

smalldragoon, 2023-04-25:
Feedback from Discord : 
- Can you check if my IP is not banned ?
- Can you check if my account is still not part of the server ( ghost account). if so, just delete it
- Do you have a Bot to moderate ? if so, can you check if I'm not part of its own blacklist ' discord tag )?
- Thanks

mdevaev, 2023-04-25:
What is your primary account?

smalldragoon, 2023-04-26:
mumphrey#3233

mdevaev, 2023-04-27:
Try again with it

mdevaev, 2023-04-27:
Your main account has been banned. Change your password, maybe it was spam.

closed 2023-04-27 by mdevaev