pikvm/pikvm #997: NGINX Config Block Connections from WAN IP Use DNS Only

Mijobar, 2023-04-19:
Hello! 

Excellent product! I have successfully port forwarded via WAN IP and am proxies requests through Cloudflare Proxy. 

My only ask is how can I prevent direct access to the Port Forwarded PiKVM via the WAN IP and only route through  'your.domain.com' ?

Would it be something like the example below?  I just don't know where to put it? Thanks! 

server {
    listen       WAN_IP_ADDRESS:80;
    server_name  your.domain.com;

    if ($host != 'your.domain.com') {
        return 444;
    }

    # other configuration here
}

mdevaev, 2023-04-19:
Hello! PiKVM uses the regular Nginx to process requests on the first level. You can find the configuration in `/etc/kvmd/nginx/nginx.conf`.

Mijobar, 2023-04-19:
Thanks! I was able to come to a solution by adding the following:
`39     server {
40
41         include /etc/kvmd/nginx/listen-http.conf;
42         include /etc/kvmd/nginx/certbot.ctx-server.conf;
43         include /etc/kvmd/nginx/redirect-to-https.conf;
44         if ($host != 'your.domain.com'){
45         return 444;}
46 }
47
48     server {
49         include /etc/kvmd/nginx/listen-https.conf;
50         include /etc/kvmd/nginx/ssl.conf;
51         include /etc/kvmd/nginx/kvmd.ctx-server.conf;
52         include /usr/share/kvmd/extras/*/nginx.ctx-server.conf;
53         if ($host != 'your.domain.com'){
54         return 444;}
55     }
56 }`

Thanks for your help!

mdevaev, 2023-04-19:
You're welcome!

closed 2023-04-19 by mdevaev