pikvm/pikvm #1000: UFW disables itself

DarkShadowxx, 2023-04-23:
**Describe the bug**
I have a setup where my Pi connects to a Wireguard VPN. I only want to manage it over VPN.
Any connection on the local LAN should be blocked.
I set up UFW and enabled it. But every few days, it disables itself and allows unauthorized access.

**To Reproduce**
1. Default PiKVM install
2. Enable write mode (rw)
3. Enable UFW (in rw mode)
4. Disable write mode (ro)

**Expected behavior**
I want ufw to stay enabled

**PiKVM info:**
  - RPi 4
  - PiKVM v3
  - Linux kernel: _Linux pikvm 5.15.68-3-rpi-ARCH #1 SMP Mon Oct 31 20:56:54 MSK 2022 armv7l GNU/Linux_

arch1mede, 2023-04-23:
I would not recommend running a firewall on this distro as this is using a RO FS, its likely the firewall is creating and using a dynamically generated file that its using, I would highly recommend you use [Tailscale](https://docs.pikvm.org/tailscale/). Then activate 2FA.

mdevaev, 2023-04-23:
Correction: there are no problems with using a firewall, but the tool for configuring it should be ready for the fact that rootfs is in read-only mode. For example, you can use iptables directly (although this is not very convenient).

Or you can check `journalctl -u ufw` too see what happened with it. Maybe this can be easily fixed.

mdevaev, 2023-04-27:
@DarkShadowxx sup?

DarkShadowxx, 2023-04-30:
It did not occur again until now. It only occurs randomly and may take a few days.

Next time it happens, I will post journalctl logs.

Maybe iptables is an option, but I will have to load the rules at boot. Is there anything like cron or iptables-persistent available?

mdevaev, 2023-05-01:
PiKVM is a regular Arch Linux, so you can use some standard recipes: https://wiki.archlinux.org/title/iptables

DarkShadowxx, 2023-05-04:
Seems like it only turns off when rebooting. But journalctl logs are empty.
What would be the easiest way to execute `ufw enable` at every startup?

closed 2023-05-04 by DarkShadowxx

mdevaev, 2023-05-05:
Got it