pikvm/pikvm #999: Inquiry about the progress of dynamically adjusting EDID ？

TelDragon, 2023-04-23:
First of all, thank you to the respected author for developing this great creation!

I also encountered black screen issues with different brands and models of bios when using them. When I was browsing through the solution, I noticed that we will be adding a feature to dynamically switch between edids later. I specifically asked when this feature is expected to be added?

[https://docs.pikvm.org/edid/#custom-edids](https://docs.pikvm.org/edid/#custom-edids)

mdevaev, 2023-04-23:
Hello. Presumably this year.

TelDragon, 2023-04-23:
> Hello. Presumably this year.

Wow~It's so exciting that you replied to the message so promptly

TelDragon, 2023-04-23:
@mdevaev 
There is also a small question that needs to be consulted.
The default design uses CSI Toshiba's TC357843 HDMI interface.
For servers or older computers, most of them still use VGA interface form.
The stupid method I used was to purchase a VGA to HDMI converter to use. Do you have any recommendations?
Or in this way, a black screen may appear during use. So should I prioritize troubleshooting the converter issue or the EDID issue?

mdevaev, 2023-04-23:
You can try this: https://aliexpress.com/item/3256801728005613.html

I suppose this is the converter issue.