pikvm/pikvm #1001: audio is not supported with linux 5.17 and higher kernels

kovalev0, 2023-04-24:
**Describe the bug**
When pikvm is connected to a computer running OS with linux kernel 5.17 and higher, the sound is not transmitted to the webrtc client.

**Desktop (please complete the following information):**
 - OS: altlinux
 - Browser chrome, mozilla

**PiKVM info:**
  - RPi 4
  - blikvm
  - edid data:
$ cat /etc/kvmd/tc358743-edid.hex 
00FFFFFFFFFFFF005262888800888888
1C150103800000780AEE91A3544C9926
0F50543FCD0001000101010101010101
010101010101011D007251D01E206E28
5500C48E2100001E8C0AD08A20E02D10
103E9600138E2100001E000000FC0054
6F73686962612D4832430A20000000FD
003B3D0F2E0F1E0A2020202020200144
02031D434E041303021211012021A23C
3D3E1F66030C00300080E2007F8C0AD0
8A20E02D10103E9600C48E210000188C
0AD08A20E02D10103E9600138E210000
189729A0D051842230509816009A0111
00001800000000000000000000000000
00000000000000000000000000000000
0000000000000000000000000000002E

**Additional context**
Starting from kernel 5.17, the SAD (short audio descriptor) descriptor check in ELD information has been added to the audio driver. Revert commit https://lore.kernel.org/all/20211202073338.1384768-1-kai.heng.feng@canonical.com/ solves the problem up to kernel 6.0. However, this does not solve the problem since kernel 6.1 - canceling this commit alone is not enough. Apparently, there is not enough additional information about the necessary SAD structures in the edid data.

kovalev0, 2023-04-25:
> However, this does not solve the problem since kernel 6.1 - canceling this commit alone is not enough

https://lore.kernel.org/all/20220922084017.25925-1-perex@perex.cz/
revert this commit to restore pikvm support

mdevaev, 2023-04-25:
I don't quite understand. Can this problem be solved in EDID? If support is removed, what is suggested to use instead?

velavokv, 2023-04-25:
> I don't quite understand. Can this problem be solved in EDID? If support is removed, what is suggested to use instead?

I understand that with the current edid populated, the emulated monitor is not properly recognized in the new kernel version (in terms of audio) and is evaluated as a device without audio support.  That is, if you remove these 2 commits from the kernel, then the check disappears and the audio is broadcast in the same mode and everything works.  According to the specification on edid version 1.3, you can find the corresponding audio description bytes and change them (most likely they are wrong).  Perhaps it is worth trying to pull them out of other edid monitors with audio support and substitute them for pikvm.

mdevaev, 2023-05-09:
Could you try this one?
```
00FFFFFFFFFFFF0031D8888800888888
1C150103800000780AEE91A3544C9926
0F50543FCD0001000101010101010101
010101010101011D007251D01E206E28
5500C48E2100001E8C0AD08A20E02D10
103E9600138E2100001E000000FC0050
694B564D0A20202020202020000000FD
003B3D0F2E0F1E0A202020202020013C
020321434E041303021211012021A23C
3D3E1F66030C00300080E2007F230907
078C0AD08A20E02D10103E9600C48E21
0000188C0AD08A20E02D10103E960013
8E210000189729A0D051842230509816
009A0111000018000000000000000000
00000000000000000000000000000000
000000000000000000000000000000F0
```

kovalev0, 2023-05-11:
> Could you try this one?
>
> ```
> 00FFFFFFFFFFFF0031D8888800888888
> 1C150103800000780AEE91A3544C9926
> 0F50543FCD0001000101010101010101
> 010101010101011D007251D01E206E28
> 5500C48E2100001E8C0AD08A20E02D10 [...]

This variant works great, thank you!

closed 2023-05-11 by kovalev0