pikvm/pikvm #1005: Add QR decoder to extract text from video

nui, 2023-04-28:
**Is your feature request related to a problem? Please describe.**
I want to export some binary data from server but OCR doesn't work well with base64 encoded data.

In order to get correct data from server I encode it to qrcode and screen capture it then use online service to extract data back.

**Describe the solution you'd like**

Add another menu to PiKVM to allow decode qr image directly from PiKVM then copy it to clipboard (alternative to text recognition)  

**Additional context**
This is example of qr generated by jupyter-lab
[image]

nui, 2023-04-28:
That would be an alternative option. A nice to have feature but I'm not sure it worth implement or not.

I could contribute this feature if you think it worth to add it btw.

mdevaev, 2023-04-28:
I will add it.

mdevaev, 2023-12-06:
In python, we can use either zbar or opencv.
* The zbar seems to be abandoned, and the worst thing is it has [unclosed CVEs](https://github.com/mchehab/zbar/issues/263).
* opencv is too heavy to install just for the sake of qr decoding.

It seems that there are no more ways to implement this with little blood. The rest of the python libraries use zbar in one way or another, or they are also abandoned.