pikvm/pikvm #1011: Apple TB/USB-C HDMI video doesn't work

gregsmith-movista, 2023-05-04:
**Describe the bug**
Attached PiKVM to MacBook Pro (x86) and MacBook Air M1. Both devices do not have a local HDMI port or USB-A port. Instead, they only provide USB-C or TB3/4 ports. I attempted to connect the PiKVM to a variety of USB-C or TB docks. None of them worked.

The Mac screen would be displayed for up to 10 seconds before going blank. The Keyboard and Mouse both continue to work thru the web interface, but the video just stays grey.

Devices attempted:
Apple USB-C Digital AV Multiport Adapter
OWC USB-C Travel Dock E, Portable 6 Port Hub
Anker USB C Hub, PowerExpand 6-in-1

**To Reproduce**
Steps to reproduce the behavior, like:
1. Attached PiKVM to Mac Book Pro (2019) or Mac Book Air M1, both experience the same issues.
2. Set video on Mac to "mirror" and set scale to be "720p, 60hz" as the web documentation indicates has the best shot of working.
3. Open PiKVM software, logged in (even updated etc).
4. Display appears for up to 10 seconds, before going grey
5. No amount of adjusting settings in web UI has any effect.
6. Placing the laptop into SafeMode (to bypass HDCP) has no effect.

**Expected behavior**
Expect WebUI to display video signal on Mac for more than a few seconds.

**Screenshots**
[image]
[image]
Video of device in opperation
https://cdn.discordapp.com/attachments/896216359963332628/1102785928256311316/PXL_20230502_023359915.TS.mp4

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Browser: Google Chrome
 - Version: 112.0.5615.138
 - VNC client (if used) (NA)

**PiKVM info:**
  - PiKVM Mini (unaltered)

**Additional context**
If this can not be addressed, then a change needs to occur on all official sites that Apple devices thru a USB-C or TB Dock/Adapter is not supported and probably wont work.

gregsmith-movista, 2023-05-04:
I am also wiling to send one of these docks out to a PiKVM dev to get them started on getting it supported.

mdevaev, 2023-05-07:
Hello again. Please try this edid:
```
00FFFFFFFFFFFF0010AC132045393639
201E0103803C22782ACD25A3574B9F27
0D5054A54B00714F8180A9C0D1C00101
010101010101023A801871382D40582C
450056502100001E000000FF00335335
475132330A2020202020000000FC0044
454C4C204432373231480A20000000FD
00384C1E5311000A2020202020200181
02031AB14F9005040302071601061112
1513141F65030C001000023A80187138
2D40582C450056502100001E011D8018
711C1620582C250056502100009E011D
007251D01E206E28550056502100001E
8C0AD08A20E02D10103E960056502100
00180000000000000000000000000000
0000000000000000000000000000004F
```

mdevaev, 2023-05-09:
Sup?

jricky31, 2023-05-10:
@mdevaev Could you share another `edid` for a MacBook Pro using Intel chip?

jricky31, 2023-05-10:
Update: Installed the  [ASUS PA248QV EDID](https://github.com/linuxhw/EDID/blob/master/Digital/ASUS/AUS2487/2B473481CAE6) and everything works good, including audio and video 1080p@50

@gregsmith-movista I think you should try this.

mdevaev, 2023-06-04:
@gregsmith-movista boop

liederivative, 2023-07-31:
Tried what @jricky31 suggested, but still having the same issue. After upgrading to Ventura 13.3.1 , no video at all.

liederivative, 2023-08-01:
Update: Installed Acer B246WL and it's working really nice. Still checking the audio, but video is stable at 1080p

arch1mede, 2023-08-03:
@SamShylla Hi, I would recommend you joining us on [Discord](https://discord.gg/bpmXfz5) so we can troubleshoot this live however, as a first step:

We would like to see the link you got the cable from, we would need to know what PiKVM hardware you are using...

"Is it necessary to always use an adaptor instead?" No BUT at this point, its unknown what is exactly causing the issue. Could be the EDID, or cable, this also might be the OS.

"works only for a while and then stops working." - Can you provide more details on when this starts and when this stops?

You might need to play with the [edid](https://docs.pikvm.org/edid/) or you might need to use a different adapter.

davydevs, 2023-11-26:
I had the EXACT same issue as the original poster with my 15" MacBook Pro 2018.  The video would work for a maximum of 10 seconds and then the bitrate would go to 0 and the picture would go grey/black and though it was not mentioned, remote keyboard and mouse control was still working.  As far as I can see from the linked video, my situation was exactly the same. I resolved this issue with a different adapter and NO CHANGES to my PiKVM.

This is the USB-C to HDMI adapter that **DID NOT WORK WELL with PiKVM**:
WALNEW 103D71

This is the adapter that **WORKS PROPERLY with PiKVM** for me:
Anker 310 USB-C Adapter (4K HDMI) (Model No: A8312 or AK-A83120A1)

I do not know why this adapter works better for me, hopefully it is not a one-off situation, but I hope that this may help someone in the future that has a similar issue.

zszszz, 2024-05-24:
Followed above and it worked, but also needed to adjust in Mac Display Settings to Optimize for PA248QV . This is the edid I tried, didn't try others

zszszz, 2024-06-01:
I think it might be adapter issue. Using pikvm 4 plus on both.

1)Intel Mac
Adapter using before: https://www.amazon.com/gp/product/B09SDBHMN5

Adapter that fixed flickering: https://www.amazon.com/gp/product/B09KV32ZBB/

2)Apple Silicon Chip Mac
Adapter using before: https://www.amazon.com/gp/product/B09SDBHMN5
I suspect plugging HDMI directly into HDMI port on Mac fixed flickering