pikvm/pikvm #1013: TESMART 16 Port KVM GPIO offline

joelidrive, 2023-05-16:
**Describe the bug**
After about 1-2 hours the GPIO buttons to switch inputs on the PiKVM are greyed out and do not work (does not switch input when pressed). This is on a 16-port TESMART switch. Model: HKS1601A10. Switch is still online and able to switch inputs through Network control on the TESMART Switch Controller. Only action that fixes the issue is by powering the TESMART KVM switch on and off again. After a power cycle the GPIO button LEDs become green and available again. Restarting kvmd service does not fix the issue either.

**To Reproduce**
Steps to reproduce the behavior, like:
1. After about 1-2 hours from a power cycle of the TESMART KVM switch, the GPIO buttons that were made available through the overide.yaml file are greyed out and do not function anymore.
2. Updating and rebooting PiKVM still has the GPIO buttons greyed out. Restarting kvmd service still has buttons greyed out.
3. Issue repeats after the TESMART KVM switch power cycles.

**Expected behavior**
GPIO buttons for switching inputs does not grey out after a couple of hours and is able to switch inputs. Stays consistently green and available unless switch is powered off or PiKVM is updated.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 11 Build 22621
 - Browser Firefox
 - Version 113.0.1
 - No VNC client used.

**PiKVM info:**
  - Raspberry Pi 4
  - PiKVM platform v3 HDMI
  - Video capture type CSI Bridge
  - KVMD version: 3.217-1
  - uStreamer version: 5.38-3
  - Linux kernel: 5.15.68-3-rpi-ARCH

**Additional context**
Unsure if this is an issue with the TESMART switch, PiKVM, or local network environment. Don't really know where to go to help solve this issue, but here. If specific logs are required let me know and I can provide.

mdevaev, 2023-06-04:
Please attach the full output of `journalctl -u kvmd`

joelidrive, 2023-06-05:
Was able to solve the issue.

It was a local environment issue.

A random PC on my network was always attempting to grab the same static IP set for the switch. Changed IP for the Tesmart switch.

I longer have this issue. Thanks for reply :)

closed 2023-06-05 by joelidrive

mdevaev, 2023-06-05:
:ok_hand:

timblaktu, 2023-08-20:
@joelidrive I wonder if you would be willing to share with me your use case and some detail and feedback about your design and control setup?

I bought one of these tesmart hdmi kvm switches and i'm looking for existing code/projects to use and contribute to before rolling my own.

I've got the 16 port version, connected to some dell poweredge servers and optiplex sff workstations and rpis and other small devices in my basement homelab. I use the switch in the lab space, but would like to automate control from a pc, and also eventually connect its output to  something like a pikvm to allow remote access.

joelidrive, 2023-08-21:
@timblaktu 

Sure! My use case is very unorthodox. It's used in a work environment to make my workflow a lot simpler. 

I use this to not go back and forth from my desk (physically speaking) when I install Vanilla Debian on multiple servers. It's really helpful when using ISO images as well. Being able to mount and unmount without physically touching servers saves even more time.

While there are more automated ways to do this, we have a specific way of doing things which makes the automated way of deployment more difficult. To avoid issues I use a PiKVM connected to the TESMART 16 Port KVM while driving the logic inputs via the GPIO pins as you mentioned :)

It works very well for what I need it to do and the bonus is that I can do that all from the comfort of my actual workstation versus the space used to setup servers.

I have not played around with remote access straight to the PiKVM using Tailscale since I already have remote access to my workstation setup. Might be something I do in the future!