pikvm/pikvm #1016: Support for changing meta, control, option keys (such as for Mac connecting to Linux or Windows)

traisjames, 2023-05-24:
Please note I am referring to US QWERTY keyboard layouts on Mac and non-Mac computers for this.

**Is your feature request related to a problem? Please describe.**
I use a Mac laptop as my daily driver and am use to that layout (On a windows keyboard I can usually switch without issue).  The three main modifier keys are in different configurations between Mac keyboards and Windows (and Chrome Books because pain and suffering feel good).
Mac:
control | option (alt) | command (or apple) | space bar | command | option
Everyone else:
control | Windows or meta | alt | space bar | alt | Windows | control
Chrome book:
control | there may or may not be a key here | alt | space bar | alt | control.

So here are some commonly used meta based commands
Undo/Cut/Copy/Paste:
- Windows: Control (key 1 in the list) + Z/X/C/V
- Mac: Command (key 3 in the list) + Z/X/C/V

Window Switcher:
- Windows: Alt (key 2 in the list) + Tab
- Mac: Command (key 3 in the list) + Tab

**Describe the solution you'd like**
While its impossible to perform a 1-to-1 mapping, for most usage, just mapping command on a Mac to control on Windows would greatly help.  

**Describe alternatives you've considered**
There are ways to remap the keys in Windows and Linux but those take a lot of configuration each time you switch from using a local keyboard to a remote Mac keyboard.  I would like to have the option to do it though the piKVM.

mdevaev, 2023-06-03:
Hello. Good point. I'll try to make something for this.

zer010gic, 2023-06-09:
If I understanding this request.  I actually don’t agree.  This is more of an issue for those that switch a lot between platforms and have a built up muscle memory.  Most Mac users intrinsically understand win key and command key are the same thing (functionally) moving that function to the  control on the local key board would make the Mac keyboard (local) to a Mac remote (remote) host respond inconsistently.  Maybe if this was a config option to swap meta key and control.   On a related note I would love to see the  option for a Mac virtual keyboard layout in the gui.

squircle12, 2023-06-27:
in the same situation for me although I'm more frustrated by the change of location for the " and @ keys. Yes I could change the keyboard layout on the windows machine - however usually these are locked down machines that you cant access to change the keyboard settings. 

So the situation I'd like is: 
Configure pikvm to have a keyboard mapping from Mac layout to windows layout.

mdevaev, 2023-08-11:
Okay, so there should be removable native layouts here. I'm merging this task with another one about keyboard localization, since they both must be implemented in one way: #611

closed 2023-08-11 by mdevaev