pikvm/pikvm #1017: Multiple monitor consolidation with a PiKVM 4+ and 4

BarryMar, 2023-05-26:
**Is your feature request related to a problem? Please describe.**
I deal with machines that have multiple monitors that I’d like to remotely access.  I use an HDMI KVM switch to handle this dual head setup right now (TinyPilot V2 atm , waiting for my PiKVM 4+ to arrive today), but Absolute Mouse Positioning without knowledge of the KVM state kinda sucks.

The issues are already largely laid out in [Issue 263](https://github.com/pikvm/pikvm/issues/263) 

**Describe the solution you'd like**
Since one Pi4 doesn’t have the bandwidth for two signals, one solution described is to use 2 PiKVM.  It would still be cheaper than commercial solutions.

Rather than multiple web browser windows, why couldn’t the first PiKVM unit act as a proxy for the data stream coming from the second unit?  This would allow for a unified interface to be presented.

Keyboard, mouse, and ATX events will still come from just one of the PiKVMs.

**Describe alternatives you've considered**
A possible alternative solution to a second HDMI source might be a device like this [URayTech HEVC H.265 H.264 HD HDMI Video Streaming Encoder HDMI to H.265 H.264 IP Video Encoder Transmitter with HTTP RTSP HLS RTMP RTMPS UDP SRT ONVIF ](https://a.co/d/6T8LgDK).

**Additional context**
This might even be an opportunity to create an even cheaper and more cut down PiKVM type - no ATX or OTG ports, just HDMI In. A PiKVM 4- , if you will, that could maybe feature match the URayTech.

mdevaev, 2023-06-03:
Hello. I consider this solution as one of the possible ones for a multimonitor configuration, but at the moment it is a low priority task. That is, yes, it is possible, but it is time-consuming to implement.

mannyking0711, 2025-02-21:
@mdevaev  Many people wants this feature!