pikvm/pikvm #1018: Pi-KVM Video Stream Issue in Web Sites

wohenipinle, 2023-05-30:
When connecting Pi-KVM in web sites, the video stream would keep appearing  and disappearing.
During this period, the usb signal would work still (the traget board could be controlled.).

Steps to reproduce the behavior, like:
1. Open PI-KVM websites
2. Turn on the controlled target board 
3. Video stream starts appear and disappear

Screen Shots:
[image]
[image]
[image]

Environment:
  - Raspberry Pi board version: RPi 4B
  - Video capture type: HDMI-CSI bridge
  - OS: Target Board (Win11 22H2), 
           Pi-KVM (Default image for Pi 4B)
  - Browser: Microsoft Edge

mdevaev, 2023-06-03:
Also try Chrome instead of Edge.

mdevaev, 2023-06-29:
Closed because there is no feedback from user.

closed 2023-06-29 by mdevaev