pikvm/pikvm #1025: Documentation for cloudflared is out of date

itlty, 2023-06-07:
Cloudflared docs state that you must build cloudflared. however Cloudflare provide the binaries already.
I suggest the Installation steps to replace those in the documentation currently
Original docs here: https://docs.pikvm.org/cloudflared/#cloudflare-tunnel-steps
### Installing Cloudflared
1. Use the following commands to install cloudflared:
- rw
- wget -O /usr/local/bin/cloudflared "$(curl -s https://api.github.com/repos/cloudflare/cloudflared/releases/latest | grep -e 'browser_download_url.*/cloudflared-linux-armhf"' | sed -e 's/[\ \":]//g' -e 's/browser_download_url//g')"
- chmod +x /usr/local/bin/cloudflared`

3. Add the service to cloudflared:

- cloudflared service install SERVICE_TOKEN_HERE`

### Updating Cloudflared
Use these commands to update the cloudflared daemon:
- rw
- cloudflared update
- systemctl restart cloudflared

solipsist01, 2023-06-13:
i did a fresh install of a v3 image.
using this installation worked, except i got the following error

```
Jun 13 08:30:22 pikvm systemd[1]: Starting cloudflared...
Jun 13 08:30:22 pikvm cloudflared[5231]: 2023-06-13T05:30:22Z INF Starting tunnel tunnelID=redacted
Jun 13 08:30:22 pikvm cloudflared[5231]: 2023-06-13T05:30:22Z INF Version 2023.5.1
Jun 13 08:30:22 pikvm cloudflared[5231]: 2023-06-13T05:30:22Z INF GOOS: linux, GOVersion: go1.19.6, GoArch: arm
Jun 13 08:30:22 pikvm cloudflared[5231]: 2023-06-13T05:30:22Z INF Settings: map[no-autoupdate:true token:*****]
Jun 13 08:30:22 pikvm cloudflared[5231]: 2023-06-13T05:30:22Z INF Generated Connector ID: redacted
Jun 13 08:30:22 pikvm cloudflared[5231]: 2023-06-13T05:30:22Z INF Initial protocol quic
Jun 13 08:30:23 pikvm cloudflared[5231]: 2023-06-13T05:30:23Z INF ICMP proxy will use 192.168.10.254 as source for IPv4
Jun 13 08:30:23 pikvm cloudflared[5231]: 2023-06-13T05:30:23Z INF ICMP proxy will use fe80::dea6:32ff:fee2:8c27 in zone eth0 as source for IPv6
Jun 13 08:30:23 pikvm cloudflared[5231]: 2023-06-13T05:30:23Z ERR Error opening metrics server listener error="lookup localhost on 1.1.1.1:53: no such host"
Jun 13 08:30:23 pikvm cloudflared[5231]: Error opening metrics server listener: lookup localhost on 1.1.1.1:53: no such host
Jun 13 08:30:23 pikvm systemd[1]: cloudflared.service: Main process exited, code=exited, status=1/FAILURE
Jun 13 08:30:23 pikvm systemd[1]: cloudflared.service: Failed with result 'exit-code'.
Jun 13 08:30:23 pikvm systemd[1]: Failed to start cloudflared.
Jun 13 08:30:28 pikvm systemd[1]: cloudflared.service: Scheduled restart job, restart counter is at 122.
Jun 13 08:30:28 pikvm systemd[1]: Stopped cloudflared.
```

after i added localhost to /etc/hosts the service would start.
note, before changing this, localhost actually resolved. (as root). 

```
# Static table lookup for hostnames.
# See hosts(5) for details.
127.0.0.1 localhost

```

mdevaev, 2023-06-13:
Please send pull request.

itlty, 2023-06-13:
Cloudflared is attempting to get the IP for the DNS name localhost from 1.1.1.1 and not from your local resolver (I see you figured that out) be that libc resolver or systemd-resolved. I am not able to reproduce this on a v4
Pls confirm the following as I am curious about the cause of this.
 - is systemd-resolved.service running? (systemctl is-active systemd-resolved.service)
 - contents of /etc/resolv.conf (cat /etc/resolv.conf)

itlty, 2023-06-13:
PR#1030 @mdevaev

solipsist01, 2023-06-13:
This is the output :)

```
[root@pikvm ~]# systemctl is-active systemd-resolved.service
active
[root@pikvm ~]# cat /etc/resolv.conf
# This is /run/systemd/resolve/resolv.conf managed by man:systemd-resolved(8).
# Do not edit.
#
# This file might be symlinked as /etc/resolv.conf. If you're looking at
# /etc/resolv.conf and seeing this text, you have followed the symlink.
#
# This is a dynamic resolv.conf file for connecting local clients directly to
# all known uplink DNS servers. This file lists all configured search domains.
#
# Third party programs should typically not access this file directly, but only
# through the symlink at /etc/resolv.conf. To manage man:resolv.conf(5) in a
# different way, replace this symlink by a static file or a different symlink.
#
# See man:systemd-resolved.service(8) for details about the supported modes of
# operation for /etc/resolv.conf.

nameserver 1.1.1.1
nameserver 8.8.8.8
search .
[root@pikvm ~]#

```

itlty, 2023-06-13:
That is correct output. Its rather interesting to me that it did not work without adding the entry for localhost in /etc/hosts.

solipsist01, 2023-06-13:
i think it's really strange as well, especially because localhost resolved fine in the terminal.
why on earth would the service try to resolve localhost from 1.1.1.1

before this morning, i had a v2 image running with usb dongle, that install used cloudflare built, from the old docs.

Yesterday my v3 kit arrived. 
i assembled the kit this morning, downloaded the v3 preassembled image. changed root password, changed login to never authenticate, cuz, ztna, and after that pasted your install.

only things that have changed are the image, and the cloudflare install.

itlty, 2023-06-13:
Did you update the OS & reboot also?
As far as I can tell that should not happen. Ive done this on a v3 in the mean time also and its fine.

solipsist01, 2023-06-13:
i didn't :)

mdevaev, 2023-06-13:
Merged.

closed 2023-06-13 by itlty