pikvm/pikvm #1028: PiKVM only connecting to FritzBox Wifi if available when booting

JsBergbau, 2023-06-08:
**Describe the bug**
PiKVM is  only connecting to FritzBox Wifi if available when booting. PiKVM is with latest image downloaded a few minutes ago for Raspberry PI Zero 2W platform.

I have a Wifi Let's call it "MySSID" on 3 AccessPoints with OpenWRT + FritzBox, so all 4 devices use the same SSID. PiKVM won't connect to any of the OpenWRT APs, although have no problem with other devices. There are other Raspberry Zero W (not 2 Ws) in the Wifi and Smartphones, so Wifi is working flawlessly.

When Fritzbox Wifi is not available PiKVM boots, but doesn't get connected to network. After enabling then Wifi on Fritzbox, even after about 5 minutes, PiKVM is not available via network. I pull the power and re-plug and it is working again. It also can't be "stuck" on one of the OpenWRT APs, because I exec `wifi` command which resets the Wifi and I see in FritzBox that the devices which were before on the OpenWRT devices roam to Fritzbox.

**To Reproduce**
See above, you need some APs.

**Expected behavior**
Even if there is a problem with OpenWRT and PiKVM (I didn't find by google though), PiKVM should detect that there is FritzBox Wifi and connect. And of course PiKVM should also work with the OpenWRT APs. 

**Screenshots**
Don't know what to screenshot here, it is a network connectivity issue.

**Desktop (please complete the following information):**
not applicable

**PiKVM info:**
  - Raspberry Pi board version [e.g. RPi 4] Pi Zero 2W
  - PiKVM platform [e.g. v2-hdmi] v2-hdmi
  - Video capture type [e.g. CSI bridge] CSI bridge
  - KVMD version: `pacman -Q | grep kvmd`
 ```
kvmd 3.228-1
kvmd-fan 0.26-1
kvmd-oled 0.23-1
kvmd-platform-v2-hdmi-zero2w 3.228-1
kvmd-webterm 0.43-1
```
  - uStreamer version: `pacman -Q | grep ustreamer` ustreamer 5.39-1
  - Linux kernel: `uname -a` Linux pikvm 5.15.68-3-rpi-ARCH #1 SMP Mon Oct 31 20:56:54 MSK 2022 armv7l GNU/Linux

**Additional context**
Took me about an hour to find out why PiKVM isn't working any more and I am so glad, that FritzBox Wifi was on while I was setting it up initially that everything worked. Otherwise it would have been much more frustrating.

arch1mede, 2023-06-09:
Hi, there are a few limiting factors that are in play as I see it.

For context, PiKVM is not its own distro, you can see this from our [FAQ](https://docs.pikvm.org/faq/)

[image]

We are unfortunatly not able to assist with Wifi/Network issues, this might be a configuration issue on your wifi router or the code running on your AP's, you might have some luck with posting your wifi issue on an [arch linux forum](https://archlinuxarm.org/forum/) or reach out to the AP support and see if they can offer any suggestions.

As a test, you could install plain Arch arm linux from the arch downloads and see if you get the same results.

I find that arch linux is more finicky than other distro's.

I have several AP's and mine are connecting.

**_I am advanced with linux, so you can help me to debug this. I tried without success. Since USB is in OTG mode, I can't connect a keyboard or a USB NIC. I tried already, but they don't work, probably because of OTG mode._**
Yes, its best to connect using the RPi standard USB to UART cable to the GPIO pins to get console access

You can also hook up a power/data cable to the first port next to the hdmi port, take the sd card out, add ENABLE_OTG_SERIAL=1 parameter to /boot/pikvm.txt, if not present, add a file and insert, put sd card back in, hook the cable to a pc, this will boot the z2w, youll need to find the com port, for me it was com3 bring up putty and connect to the serial port 115200 and login, then you should be able to run some iw commands

### Common wifi commands
- `iwconfig` manipulate the basic wireless parameters
- `iwlist`   allow's you to initiate scanning and list frequencies, bit-rates, encryption keys…
- `iwspy`    displays per node link quality
- `iwpriv`   allow's you to manipulate the Wireless Extensions specific to a driver (private)

- Some examples
```
iw dev wlan0 scan | egrep "signal:|SSID:" | sed -e "s/\tsignal: //" -e "s/\tSSID: //" | awk '{ORS = (NR % 2 == 0)? "\n" : " "; print}' | sort
```
```
iwlist wlan0 scan | egrep "Cell|ESSID|Signal|Rates"
```
```
iwlist wlan0 scan
```
```
iw wlan0 info
```

arch1mede, 2023-06-09:
Just a quick followup:
I broke out my ZeroW2  for this issue to test and while I can agree that it is difficult and frustrating to get wifi working, I had to reflash the sd card, and use a different wifi AP to connect due to not being able to connect to my main wifi AP. It would connect but refused to give me an ip, I would need to run `dhclient -v` everytime.

The rule of thumb with wifi
If you have a modded AP, make sure and also employ a default AP with 0 modifications. While I can agree that modding an AP is fun, at first, in the long run, it comes down to time VS patience.

JsBergbau, 2023-06-10:
Thanks for the trick with the serial console via usb. That helps a lot. 

So now I have access via serial console. I can scan the Wifi and find all the wifis from 2 of the 3 OpenWRT APs.

Tried a lot and finally found how I get it working with the OpenWRT-APs.
They are configured WPA2+WAP3 mixed mode `option encryption 'sae-mixed'`

After changing 
/etc/wpa_supplicant/wpa_supplicant-wlan0.conf

to

```
ctrl_interface=/run/wpa_supplicant
update_config=1

network={
        ssid="MySSID"
        #psk="XXX"
        psk=XXX
		key_mgmt=WPA-PSK-SHA256
}
```
It finally works.

`ctrl_interface=/run/wpa_supplicant`
Is very helpful so you can connect to wpa_cli and I have seen 
`CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="XXX" auth_failures=5 duration=104 reason=CONN_FAILED`

Fritzbox is also configured to use WPA2+WPA3
Don't know why it didn't work with PiKVM and OpenWRT.

To avoid this problem I suggest you to change generation of /etc/wpa_supplicant/wpa_supplicant-wlan0.conf to generate 2 entries. 

```
ctrl_interface=/run/wpa_supplicant
update_config=1

network={
        ssid="XXXSSID"
        #psk="pskClear"
        psk=pskHex
		key_mgmt=WPA-PSK-SHA256
}

 network={
      ssid="XXXSSID"
      #psk="pskClear"
      psk=pskHex
}
```
If finally tested, with this it is possible to use FritzBox and OpenWRT. I also recommend using `ctrl_interface=/run/wpa_supplicant` because this enables usage of wpa_cli out of the box.

PS: Can you tell me how to restart wpa_supplicant under arch linux? I had to reboot because `systemctl restart wpa_supplicant` and `systemctl restart dhcpcd` didn't work to use updated config.

mdevaev, 2023-07-10:
I think systemctl restart `systemd-networkd` should work

sorcodiv, 2023-07-31:
Hi, it looks like "AP-STA-POSSIBLE-PSK-MISMATCH" problem with Raspberry PI and OpenWRT.
Do you have error in OpenWRT System Log "... daemon.notice hostapd: wlan0: AP-STA-POSSIBLE-PSK-MISMATCH ..."?
Some useful information you can find here:
https://forums.raspberrypi.com/viewtopic.php?t=327256
https://forums.raspberrypi.com/viewtopic.php?t=269221

The solution for Raspios is simple: you should downgrade firmware version. Download and isntall "firmware-brcm80211_20190114-1+rpt4_all.deb" from here: http://archive.raspberrypi.org/debian/pool/main/f/firmware-nonfree/
`dpkg -i firmware-brcm80211_20190114-1+rpt4_all.deb`

For Arch and Raspberry Pi 4 you can fix the problem manually:
- unpack files "brcmfmac43455-sdio.bin", "brcmfmac43455-sdio.clm_blob" and "brcmfmac43455-sdio.txt" from "firmware-brcm80211_20190114-1+rpt4_all.deb";
- mkdir "fix" on root of sdcard FAT partition and copy "brcmfmac43455-sdio.*" to this dir;
- replace system files with script:
```
cp /boot/fix/brcmfmac43455-sdio.txt /lib/firmware/brcm/brcmfmac43455-sdio.txt
cp /boot/fix/brcmfmac43455-sdio.bin /lib/firmware/brcm/brcmfmac43455-sdio.bin
cp /boot/fix/brcmfmac43455-sdio.clm_blob /lib/firmware/brcm/brcmfmac43455-sdio.clm_blob
cp /boot/fix/brcmfmac43455-sdio.txt /lib/firmware/updates/brcm/brcmfmac43455-sdio.txt
cp /boot/fix/brcmfmac43455-sdio.bin /lib/firmware/updates/brcm/brcmfmac43455-sdio.bin
cp /boot/fix/brcmfmac43455-sdio.clm_blob /lib/firmware/updates/brcm/brcmfmac43455-sdio.clm_blob
cp /boot/fix/brcmfmac43455-sdio.bin /lib/firmware/cypress/cyfmac43455-sdio.bin
cp /boot/fix/brcmfmac43455-sdio.clm_blob /lib/firmware/cypress/cyfmac43455-sdio.clm_blob
```
- reboot.
If you want to do this on installation, just make sh-script in /pikvm-scripts.d/ folder on sdcard FAT partition.

mdevaev, 2023-08-01:
@sorcodiv Thank you for this. I think the issue can be closed now.

JsBergbau, 2023-08-01:
Shouldn't be the downgraded firmware shipped by the latest PiKVM Images before closing this issue?

mdevaev, 2023-08-01:
This is a very, very rare issue, you are the first one who has encountered this. A downgrade may break something for other users.