pikvm/pikvm #1033: Arudino HID: Program error "obsolete use of -b <clock> option for bit clock; use -B <clock>"

jcormier, 2023-06-19:
**Describe the bug**
Programming an Arduino SparkFun Pro Micro 3v3 following the instructions:
https://docs.pikvm.org/arduino_hid/#ps2-keyboard
https://docs.pikvm.org/flashing_hid/#spi-firmware

I've attached the SS to the RX LED as mentioned in the note. As well as connecting MOSI/MISO/SCLK to pins 16/14/15. As well as the reset with a transistor.
https://images.theengineeringprojects.com/image/webp/2020/12/Introduction-to-Arduino-Pro-Micro-2.png.webp?ssl=1

I've updated the platformio file "board = sparkfun_promicro8".

Is the "obsolete use of -b <clock>" warning relevant?  The 3.3V part runs slower than the normal Micro, so perhaps the spi clock is set too high?
Where would I change this in the hid project?

```
platformio run --environment 'spi' --project-conf 'platformio-avr.ini' --target upload
Processing spi (platform: atmelavr; board: sparkfun_promicro8; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/sparkfun_promicro8.html
PLATFORM: Atmel AVR (4.2.0) > SparkFun Pro Micro 3.3V/8MHz
HARDWARE: ATMEGA32U4 8MHz, 2.50KB RAM, 28KB Flash
DEBUG: Current (simavr) External (simavr)
PACKAGES: 
 - framework-arduino-avr @ 5.1.0 
 - tool-avrdude @ 1.70100.0 (7.1.0) 
 - toolchain-atmelavr @ 1.70300.191015 (7.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 11 compatible libraries
Scanning dependencies...
Dependency Graph
|-- HID-Project @ 2.8.2+sha.afae388
|   |-- HID @ 1.0
|-- ps2dev @ 0.0.3+sha.a043002
|-- digitalWriteFast @ 1.0.0
|-- HID @ 1.0
|-- drivers-avr
|   |-- drivers
|   |   |-- digitalWriteFast @ 1.0.0
|   |-- ps2dev @ 0.0.3+sha.a043002
|   |-- HID-Project @ 2.8.2+sha.afae388
|   |   |-- HID @ 1.0
|-- drivers
|   |-- digitalWriteFast @ 1.0.0
Building in release mode
Checking size .pio/build/spi/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  15.6% (used 399 bytes from 2560 bytes)
Flash: [====      ]  36.5% (used 10464 bytes from 28672 bytes)
Configuring upload protocol...
AVAILABLE: custom
CURRENT: upload_protocol = custom
Uploading .pio/build/spi/firmware.hex
avrdude warning: obsolete use of -b <clock> option for bit clock; use -B <clock>

avrdude error: AVR device not responding
avrdude error: initialization failed, rc=-2
        the programmer ISP clock is too fast for the target
        - use -B to set lower ISP clock frequency, e.g. -B 125kHz
        - use -F to override this check

avrdude done.  Thank you.

*** [upload] Error 1
```

mdevaev, 2023-06-21:
Hello. You can find the config in `avrdude-rpi.conf`:
```
programmer
id = "rpi";
    desc = "RPi SPI programmer";
    type = "linuxspi";
    reset = 25;
    baudrate = 400000;
;
```
You can play with `baudrate` option I think. Try to reduce it by half, but I am not sure which configuration will be correct since I have not encountered this.

jcormier, 2023-06-21:
Thanks.  I get the same error when changing the baudrate.  Is there a way to update the code so it uses the correct -B argument?

mdevaev, 2023-06-21:
> Is there a way to update the code so it uses the correct -B argument?

This is a warning, not a critical error. The option is still working.

> Additionally is there a way to program the device over USB instead of over SPI.

Edit `platformio-avr.ini` and replace `upload_*` options in `_common_spi` to single `upload_port = /dev/ttyACM0`.

PS: I don't know how to make it in arduino ide, I'm not using it.

mdevaev, 2023-06-29:
Hi. How it's going?

jcormier, 2023-06-29:
Hi, yeah sorry was on vacation and am trying to catch up on other projects.  I hope to get back to this soon. Thanks for checking in.

mdevaev, 2023-08-08:
Sup?

jcormier, 2023-08-09:
My TODO list just keeps getting longer xD.  Its still setup on my desk, I just need a few hours of breathing room to get back to it...

mdevaev, 2023-08-09:
What do you think about making the HID on Raspberry Pico?

mdevaev, 2023-10-07:
Ping

mdevaev, 2023-10-08:
So, I'm closing it. Please contact support in discord if you want to continue this.

closed 2023-10-08 by mdevaev

jcormier, 2023-10-09:
> So, I'm closing it. Please contact support in discord if you want to continue this.

Thanks, that fair. 

> What do you think about making the HID on Raspberry Pico?
I do have some Picos at home.  Did someone get one of those working as a HID?

mdevaev, 2023-10-09:
Ofc. This solution works, but has not yet been officially released. Are you using Raspberry 3?

jcormier, 2023-10-09:
I'm not sure which PI this is.  The case doesn't mention a version and the cape is blocking access to the PI.  The cape is labeled "PI-KVM Shield v3.3"

mdevaev, 2023-10-09:
Take a photo

jcormier, 2023-10-18:
I got it working with the raspberry pico w. Posted in the discord. Yay

mdevaev, 2023-10-19:
Bueno :)