pikvm/pikvm #1035: failed retrieving file 'pikvm.db' from files.pikvm.org

Zitt, 2023-06-21:
**Describe the bug**
pikvm.db failed to download
error: failed retrieving file 'pikvm.db' from files.pikvm.org : Resolving timed out after 10000 milliseconds
error: failed to synchronize all databases (download library error

**To Reproduce**
Steps to reproduce the behavior, like:
rw
pacman -Sy  nfs-utils

**Expected behavior**
All the dbs seem to pass except pikvm.org

**PiKVM info:**
  - Raspberry Pi 4 Computer Model B 8GB
  - PiKVM V3 
  - CSI
  - KVMD version: 
kvmd 3.227-1
kvmd-fan 0.26-1
kvmd-oled 0.23-1
kvmd-platform-v3-hdmi-rpi4 3.227-1
kvmd-webterm 0.43-1
  - uStreamer version: ustreamer 5.39-1
  - Linux kernel: Linux [redacted] 5.15.68-3-rpi-ARCH #1 SMP Mon Oct 31 20:56:54 MSK 2022 armv7l GNU/Linux
 
**Additional context**
suspect this may be related to our corp firewall wrecking havoc. Not able to connect to pikvm.org at all. A message to our IT department resulted in:
"It appears to be a DNS issue specific to this site. "
and it briefly worked ... but now fails to connect again.

flpydsk, 2023-06-22:
This sounds like its specific to you and not going to be an issue with pikvm
Run the following:
nslookup files.pikvm.org
nslookup files.pikvm.org 1.1.1.1 #(You must be able to reach 1.1.1.1, if you cant try google @ 8.8.8.8)

mdevaev, 2023-06-22:
Everything is working. Check your network configuration.

closed 2023-06-22 by mdevaev

lesdjins, 2024-11-03:
How did you fix this?