pikvm/pikvm #1038: Agent / Interface

Mightyretro, 2023-06-23:
Hi guys hope your doing well.

is there possible to have like implement something similar to interface and maybe host it yourself x86 machine if you have bunch of lab computers or friends or familys computer need remote management but still use (Pikvm x86 --->> friends house Pikvm arm)

https://github.com/amidaware/tacticalrmm

https://rport.io/

Kind Regards // Mighty

mdevaev, 2023-06-29:
Hi. Unfortunately, I didn't understand anything from your request. What exactly do you want to get?

mdevaev, 2023-07-10:
Sup?

Mightyretro, 2023-07-14:
Sorry for late answer ;)

1. is it possible to update interface a bit something similar to down below its mutch easier categorise vm or computer you connect to for example "Homelab" etc
https://github.com/amidaware/tacticalrmm
https://rport.io/

2. is it possible to run pikvm on your own x86 hardware and you perhaps install an "Agent on client side" or also Pikvm / BliKVM PCIe...?

Kind Regards // Mighty

closed 2023-12-03 by mdevaev