pikvm/pikvm #1039: The ATX menu modification to allow 2 Power and 2 Reset items

mircevski, 2023-06-24:
Hi all, 

Initially I've asked this question on the #support Discord channel, but I was advised to post here if I do not get any answer ... which I did not get after all.

So, the question is: Is there a way to modify the ATX menu to have more commands which will correspond to other pins then the Power & Reset ?
Basically, what I’d like to accomplish is to clone the commands Power (short/long) & Reset which are being sent to pins, let’s say, 1-2 & 3-4 and have the same commands sent to pins 5-6 & 7-8 … by this I will lose the power led and hdd led, but I’ll gain the ability to power&reset a secondary machine
I’ve read that the way to customize things is to edit the /etc/kvmd/override.yml … shall I just add a copy of the ATX section in main.yml and modify that ?
In theory, I should replace the ATX section from here …
[image]

… with this in override.yml
[image]

But it seems incomplete
… since the menu itself won’t be modified
I’m using PiKVM v4 mini which connects to an Ezcoo KVM switch
I was able to solve the problem of remotely switching between inputs, but my last remaining part is the actual power controls
I need to sacrifice the Power Led and Hdd Led to accomolish this which is fine … I just need to understand where to modify the code

Any help ?
Thanks

mdevaev, 2023-07-10:
Answered, I guess. Please feel free to ping me if not, we'll reopen it.

closed 2023-07-10 by mdevaev

mircevski, 2023-07-11:
Hi, did what you sugested, so I can control 2 machines by reusing the reset as power for the second one.

But, how about the others (power led, hdd led) ? Are they hardcoded in any way ? Or their functionalities are programable ?

Kind regards,

Trimis de pe Outlook pentru iOS<https://aka.ms/o0ukef>
________________________________
De la: Maxim Devaev ***@***.***>
Trimis: Tuesday, July 11, 2023 12:46:38 AM
Către: pikvm/pikvm ***@***.***>
Cc: mircevski ***@***.***>; Author ***@***.***>
Subiect: Re: [pikvm/pikvm] The ATX menu modification to allow 2 Power and 2 Reset items (Issue #1039)

Answered, I guess. Please feel free to ping me if not, we'll reopen it.

mdevaev, 2023-07-11:
These are inputs, their role cannot be changedю

mircevski, 2023-07-11:
Got it, thanks

Trimis de pe Outlook pentru iOS<https://aka.ms/o0ukef>
________________________________
De la: Maxim Devaev ***@***.***>
Trimis: Tuesday, July 11, 2023 1:00:17 PM
Către: pikvm/pikvm ***@***.***>
Cc: mircevski ***@***.***>; Author ***@***.***>
Subiect: Re: [pikvm/pikvm] The ATX menu modification to allow 2 Power and 2 Reset items (Issue #1039)

These are inputs, their role cannot be changed