pikvm/pikvm #1040: Audio - mic input - again ;)

JustinGuese, 2023-06-26:
**Is your feature request related to a problem? Please describe.**

Video calls on controlled device

**Describe the solution you'd like*

I've already read that many people say it's hard, which I can understand for using the audio jack... But can't we connect to Bluetooth using the raspi's Bluetooth module? And then have maybe even another browser window to forward audio like in an online call? 

**Describe alternatives you've considered**

Audio seems to be an issue for a lot of people. Forwarding the microphone is one of the most important features for me.

**Additional context**
Add any other context or screenshots about the feature request here.

837, 2023-10-28:
I would pay to get this!

semi0202, 2024-03-24:
guys, please prioritize work on this feature, this is extreamly important for many users and it completes pikvm as a product to remote control the host.

plia7, 2024-05-26:
@mdevaev Is this something that can be used:

https://github.com/jacktrip-webrtc/jacktrip-webrtc

Thanks.

victoremerick, 2024-07-05:
another way to do that is create a support to simulate a microphone and headset over USB, but i don't have idea to do that.

zhangyoufu, 2024-07-05:
```
cd /sys/kernel/config/usb_gadget/kvmd
echo '' >UDC
mkdir -p functions/uac2.usb0
ln -s functions/uac2.usb0 configs/c.1/
echo fe980000.usb >UDC
```

USB UAC2 works smoothly with my Windows 10 box. It's possible to adjust number of channels and sample rate by the way.

victoremerick, 2024-07-10:
I'm trying to use my headset from windows to connect on Mac Mini, because on mac mini I need to make some calls. I only need to use 1 mouse, 1 keyboard and 1 headset (audio output I can use the HDMI port, but Microphone i didn't have idea).

If I use this code I will be able to use my headset remotely?

mdevaev, 2025-01-20:
Official [PiKVM V3 and V4 devices](https://pikvm.org/buy) gets two-way audio with microphone.

Now you can use voice applications remotely, or you can just emotionally tell the server what you think about it.

**https://docs.pikvm.org/audio**

*PS: I know that many people wanted this feature. I spent all the New Year holidays on it, so if you found PiKVM useful and finally got what you wanted, you can support my work on PiKVM or just give me a beer with donation to [Patreon](https://www.patreon.com/pikvm) or [Paypal](https://paypal.me/pikvm). Happy PiKVMing!*

closed 2025-01-20 by mdevaev