pikvm/pikvm #1043: Scroll distance for mouse wheel

visit1985, 2023-06-30:
I'm using PiKVM for quite some time now. And I'm pretty happy  with it.

I own a Logitech MX Master 2S Mouse, which has this SmartShift feature which allows you to scroll short distances with kind of a "snap in" haptic feedback after around 3mm of finger movement.

The problem is, that PiKVM reacts after around 2-3 snaps to the scroll action, which sometimes leads to a page jump on the viewed document. Is there a way to influence this?

The "Mouse polling rate" setting seems to have no effect on it.

If you need any further information, I can help you debug it.

Thanks in advance!

mdevaev, 2023-07-01:
Mouse scrolling is a pain. Try to play with this options:
[image]
In my case, the config above works acceptable.

Which browser and OS (on both sides) you use?

visit1985, 2023-07-01:
Thanks for the hint.

> Which browser and OS (on both sides) you use?

Google Chrome (macOS Monterey) --> PiKVM --> Google Chrome (Windows 10) 

I had to upgrade to find this option in the interface. Didn't do that for more than a year. 😄 The system is just so stable, I didn't feel the need to do it.

A quick test shows improvement. I will see how it behaves during normal workdays.

closed 2023-07-01 by visit1985

mdevaev, 2023-07-02:
:ok_hand: