pikvm/pikvm #1047: Call to action for fan failure doesn't tell you where to look

voltagex, 2023-07-03:
What log, where?

[image]

This is a web interface, why not make this a clickable link to the log?

voltagex, 2023-07-03:
Yes

mdevaev, 2023-07-03:
Edit the fan config `/etc/kvmd/fan.ini` and set `idle` and `low` values from 27 to 30, for example, then restart the service: `systemctl restart kvmd-fan`.

Don't forget to remount `ro`.

voltagex, 2023-07-04:
> Edit the fan config `/etc/kvmd/fan.ini` and set `idle` and `low` values from 27 to 30, for example, then restart the service: `systemctl restart kvmd-fan`.
> 
> 
> 
> Don't forget to remount `ro`.

Sure, that fixes the underlying issue but it doesn't improve the UI.

Thanks though.

mdevaev, 2023-07-04:
Great! I will add a link to the log.

mdevaev, 2023-07-07:
I've added the link to the log in this dropdown. Solved, I guess

closed 2023-07-07 by mdevaev