pikvm/pikvm #1048: When shutting down pikvm halted message not showing anymore

plia7, 2023-07-03:
**Describe the bug**
When shutting down pikvm halted message not showing anymore

**To Reproduce**
Steps to reproduce the behavior, like:
1. Run sudo shutdown -h now

**Expected behavior**
Halted message should show on pikvm physical display.

**Actual behavior**
It seem to be stuck on the last display info it was showing when you shutdown. I.e.: ip/mem/cpu info or uptime, temp info.

**Screenshots**
n/a

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]
 - VNC client (if used)
n/a Issuing the command from ssh.

**PiKVM info:**
  - Raspberry Pi board version [e.g. RPi 4]
v3 preassembled
  - PiKVM platform [e.g. v2-hdmi]
  - Video capture type [e.g. CSI bridge]
  - KVMD version: `pacman -Q | grep kvmd`
3.231-1
  - uStreamer version: `pacman -Q | grep ustreamer`
  - Linux kernel: `uname -a`

**Additional context**
For reference, in 3.169 it used to show "halted".

plia7, 2023-07-04:
@mdevaev I flashed a new micro sd card with the latest image
I updated, set passwords, enable VNC, install tailscale, update configs as needed
still able to reproduce the same issue.

mdevaev, 2023-07-05:
I'll check it

kkevin6154, 2023-07-05:
@plia7 - Thank you for logging this issue.  
@mdevaev - I'm also using the v3 preassembled and can confirm that the "halted" message is no longer displayed on the OLED screen when issuing the "shutdown" command in the terminal after updating from Linux 5.15.68-2-rpi-ARCH to Linux 5.15.68-3-rpi-ARCH.

Thank you!  :-)

kkevin6154, 2023-07-05:
To narrow this down further, I've confirmed that the halted message displayed as expected on the kvmd version 3.219-1 (20230525) and stopped working on the version after this of kvmd 3.227-1 (20230605).  Hope this helps!  Thanks!

mdevaev, 2023-07-07:
I suppose this is related with systemd update, but I didn't understand yet how to fix it.

mdevaev, 2023-07-08:
Fixed. Update OS.

closed 2023-07-08 by mdevaev

kkevin6154, 2023-07-12:
Confirmed resolved.  Thank you!