pikvm/pikvm #1049: Memory leak leading to performance slowdown/reboot when using Jump Desktop app from an iPhone

plia7, 2023-07-04:
**Describe the bug**
Memory leak leading to performance slowdown/reboot when using Jump Desktop app from an iPhone

**To Reproduce**
1. Open Jump Desktop app on your iPhone
2. Connect to your pikvm vnc server
3. Leave the app open, multi task, switch to other apps or let the phone lock the phone/screen to turn off
4. Occasionally you could come back to the app and do some actions clicking/typing/etc - Make sure to go back to the app sometimes so it doesn't time out and disconnect (or the growing will stop).
5. Run periodically htop command from pikvm ssh terminal, sort by mem % and note it.

**Expected behavior**
kvmd-vnd should not reach 15% mem or more
the total memory should not grow beyond the baseline of ~200-450M/1.77G

**Actual behavior**
the total memory keeps growing to 1.22G/1.77G and beyond leading to performance slowdown and eventual reboot
kvmd-vnd is reaching 25% mem and beyond

**Screenshots + full logs**
If applicable, add screenshots to help explain your problem.
[image]
[image]
[image]

See full htop and log by time stamp events, starts at 4:52pm, at 5:15pm it's halting and then at 5:25pm the pikvm is back from the halting state + journalctl command log below it:

[htop and journalctl - censored.txt](https://github.com/pikvm/pikvm/files/11943238/htop.and.journalctl.-.censored.txt)

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 iOS
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]
 - VNC client (if used)
 Jump Desktop v9.7.8 for iOS

**PiKVM info:**
  - Raspberry Pi board version [e.g. RPi 4]
  - PiKVM platform [e.g. v2-hdmi]
  v3 preassembled
  - Video capture type [e.g. CSI bridge]
  - HDMI
  - KVMD version: `pacman -Q | grep kvmd`
  3.231-1
  - uStreamer version: `pacman -Q | grep ustreamer`
  -5.41-1
  - Linux kernel: `uname -a`
Linux pikvm 5.15.68-3-rpi-ARCH #1 SMP Mon Oct 31 20:56:54 MSK 2022 armv7l GNU/Linux

**Additional context**
Add any other context about the problem here.

plia7, 2023-07-04:
I flashed a new micro sd card with the latest image
I updated, set passwords, enable VNC, install tailscale, update configs as needed
still able to reproduce the same issue.

mdevaev, 2023-07-05:
Fixed in KVMD 3.232.

closed 2023-07-05 by mdevaev