pikvm/pikvm #1050: No local cursor is being shown when connecting via vnc and the remote cursor should be hidden

darkbasic, 2023-07-04:
**Describe the bug**
When connecting to my PiKVM v4 via tigervnc using [these settings](https://docs.pikvm.org/vnc/#enabling-vnc-on-the-pikvm-side) no local cursor is being shown, instead it shows the remote cursor (which lags too much to be usable).
The PiKVM web UI shows both the local cursor (a circle) and the remote cursor. Ideally the remote cursor should be hidden, not sure how but maybe by faking the input events as touch inputs so that the remote cursor won't even move until you click.
If the remove cursor can indeed be hidden then maybe the local one [should be changed](https://github.com/pikvm/pikvm/issues/548) to something more visible than the thin circle in the web UI.

I see this as two separate issues:
1) The local cursor should absolutely be shown in tigervnc to make it usable and on par with the web UI.
2) The remote cursor should be hidden completely to make it actually pleasant to use. The web ui right now is "barely" usable because having two cursor is confusing and the remote one lags behind the local one (even when setting the polling interval to 10 ms).

The first one might be a tigervnc bug (I've already filed a bug report), so I would concentrate on the second one: hiding the remote cursor. How can we achieve this? Maybe by faking the inputs as touch inputs?

**To Reproduce**
1. Connect to a PiKVM v4 using [these settings](https://docs.pikvm.org/vnc/#enabling-vnc-on-the-pikvm-side)
2. Remote cursor is being shown (laggy, gets clipped when you reach the border of the window). Local cursor is not being shown.

**Expected behavior**
The local cursor should be rendered (like the small circle shown in the PiKVM web UI).
Ideally the remote cursor should not be shown at all.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Gentoo Linux ppc64le
 - Browser Chromium
 - Version 114
 - VNC client: tigervnc 1.13.1-r3

**PiKVM info:**
  - Raspberry Pi board version: PiKVM v4 Mini
  - PiKVM platform: PiKVM v4 Mini
  - Video capture type [e.g. CSI bridge]
  - KVMD version: 3.231-1
  - uStreamer version: 5.41-1
  - Linux kernel: 5.15.68-3-rpi-ARCH

**Additional context**
Tigervnc issue: https://github.com/TigerVNC/tigervnc/issues/1642

mdevaev, 2023-07-05:
Hello! There are a few things regarding this.
* It is impossible to hide the remote cursor because PiKVM cannot affect the target OS in any way. She thinks that an ordinary mouse is connected. Emulating touch events is possible, but it will lead to worse compatibility because not all devices support touch.
* The showing of the local cursor depends entirely on the VNC client, we also cannot influence this from the VNC server inside PiKVM. In my case, I use TigerVNC on Linux and there is no local cursor.
*  It is impossible to completely get rid of the latency, although I'm working to reduce it. The reason is that this is a hardware video capture, it is not the same as a regular VNC or TeamViewer. Image capture takes time, video encoding too, [the browser also adds some delay](https://github.com/pikvm/pikvm/issues/784). This problem concerns absolutely all existing KVM over IP.
* **As for the web interface, I could add an option that hides the blue dot.** However, I suggest you try to get used to it, because it's harder to use the mouse without it. You can focus on the blue dot when you make clicks, because as soon as you make a click, the remote cursor will be instantly moved to the position of the blue dot and the click will be executed in the correct position.

darkbasic, 2023-07-05:
> Emulating touch events is possible, but it will lead to worse compatibility because not all devices support touch.

Not necessarily: we could emulate **both** mouse and touch and add a switch in the web ui to choose which one you want to use. Than means being able to emulate the mouse while in the bios and switching to touch once you booted your OS.

> The showing of the local cursor depends entirely on the VNC client, we also cannot influence this from the VNC server inside PiKVM. In my case, I use TigerVNC on Linux and there is no local cursor.

That part was a bug and there is an easy workaround: https://github.com/TigerVNC/tigervnc/issues/1340

> It is impossible to completely get rid of the latency, although I'm working to reduce it. The reason is that this is a hardware video capture, it is not the same as a regular VNC or TeamViewer. Image capture takes time, video encoding too, https://github.com/pikvm/pikvm/issues/784. This problem concerns absolutely all existing KVM over IP.

Yeah I know, I'm not complaining about the latency but rather the impossibility of hiding the remote cursor somehow.

> As for the web interface, I could add an option that hides the blue dot. However, I suggest you try to get used to it, because it's harder to use the mouse without it. You can focus on the blue dot [...]

That wouldn't help. I'd like to hide the remote cursor, not the local one! I think this can only be achieved via touch events.

mdevaev, 2023-07-05:
Got it. However, using a touch device has far-reaching consequences, for example, the inability to turn off the device in real time (due to bugs in the kernel) and a limited number of USB endpoints.

Actually, I have an idea how to solve it. We are planning one secret feature here, with the help of which your case can also be solved. So I ask you to wait a couple of months before the release.

darkbasic, 2023-07-05:
> Actually, I have an idea how to solve it. We are planning one secret feature here, with the help of which your case can also be solved.

Oh nice, curiosity is killing me know :)

> So I ask you to wait a couple of months before the release.

Sure.

mdevaev, 2023-07-05:
:ok_hand:

lyonzy, 2023-09-19:
A Windows-specific option (until we get this intriguing new feature above!) is to set all your cursors in Control Panel to `invisible.cur` which you can download from various sites (e.g. [here](http://www.rw-designer.com/cursor-detail/23254))

mdevaev, 2023-09-19:
Thanks for the info!

darkbasic, 2024-02-17:
@mdevaev any news on this?

mdevaev, 2024-02-17:
Not yet, sorry. There are a lot of tasks on other fronts

dreamsea503, 2024-03-13:
Hello. 

I am trying to use KVM to control another laptop. But my actual pointer is different from remote cursor. It's the same in the middle of the screen. But when it goes to the side, there becomes different.

Can you describe why it's happening and how to fix this issue?

My captured behavior is attached via video.

[screen-capture (1).webm](https://github.com/pikvm/pikvm/assets/108384560/ee7e1b8d-87e4-4cdf-a61e-eb22285610d8)

arch1mede, 2024-08-14:
This is for anyone else running into this same issue.

The reason for this is that the Display and Active display does not match. Windows sometimes has a display mismatch, don't know why.

The fix is to have Desktop mode (Display) and Active signal mode match.

In this example, this is a v4 that can support 1920x1200@60hz, otherwise set for standard 1080p@60hz OR whatever  resolution you want.

[image]

System ->Display->Advanced Display, then display properties, then list all modes, then choose the same display size, for Win10 this might be a little different but should generally be the same.

duyingsoft, 2024-12-24:
> Got it. However, using a touch device has far-reaching consequences, for example, the inability to turn off the virtual USB device in real time (due to bugs in the kernel) and a limited number of USB endpoints.
> 
> Actually, I have an idea how to solve it. We are planning one secret feature here, with the help of which your case can also be solved. So I ask you to wait a couple of months before the release.

hi，Do you have any new updates on this feature？ @mdevaev

darkbasic, 2024-12-24:
I think it's due time to lift the secret :)

adam083b, 2025-03-21:
Hello,

I wanted to share my solution regarding this topic, I was very annoyed by the host remote cursor even focusing on the blue dot it felt very unpleasant. As mentioned pikvm can't remove the cursor by itself so the solution is to hide the cursor directly in the host system. I've no choice but to use windows so I focused on finding a solution for it, but if you know a clean solution for linux systems it would be nice to share too. Using AutoHotKey https://www.autohotkey.com/ seems to be the most pratical solution. Install the application. then rename the following file with the extension ahk [AutoHideCursor.txt](https://github.com/user-attachments/files/19377446/AutoHideCursor.txt) You just need to double click on the file to run it, use the shortcut win+c to hide the cursor and the same shortcut to show it again. @mdevaev if that's worth it you can also give the information on the official documentation

Thanks