pikvm/pikvm #1053: Orange Pi Port

HiDef888, 2023-07-09:
Any more thoughts on an Orange Pi port?
[http://www.orangepi.org/html/hardWare/computerAndMicrocontrollers/details/Orange-Pi-5-plus.html](url)

The 5 Plus now has a HDMI in - which may even simplify the build process.  An all-in-one SBC for PiKVM possibly?

mdevaev, 2023-07-09:
Hello. We don't plan to make official support for any devices other than Raspberry, but we don't mind accepting patches, provided that it will not break Raspberry. The reasons lie in the much worse support of the boards from the manufacturer, for PiKVM this is sensitive because it requires excellent operation of low-level devices such as a USB controller and a video encoder.

mdevaev, 2023-07-10:
We can continue in #unofficial_ports channel in Discord.

closed 2023-07-10 by mdevaev