pikvm/pikvm #1054: UEFI Stream is glitchy and KB+Mouse is not working

NalertN, 2023-07-12:
**Describe the bug**
Hello,
I have some Issues in the UEFI BIOS in the HP Elitebook 640 G9. The Stream is glitchy (Screenshot below) and mouse and keyboard dont work. I am not sure how to solve this Problem.

**To Reproduce**
Steps to reproduce the behavior, like:
1. Plug in the Laptop via HDMI
2. Open the BIOS settings
3. Close the Laptop so the Screen gets streamed to HDMI Output.
4. Open PiKVM Stream

**Expected behavior**
I would have expected that the Keyboard and mouse work.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Browser: tested Chrome, Edge and Firefox
 - Version: 114

**PiKVM info:**
  - Raspberry Pi board version: RPi 4
  - PiKVM platform: v3
  - Video capture type: PiKVM Shield
  - KVMD version: 3.234-1
  - uStreamer version: 5.41-1
  - Linux kernel: Linux pikvm 5.15.68-3-rpi-ARCH #1 SMP Mon Oct 31 20:56:54 MSK 2022 armv7l GNU/Linux

mdevaev, 2023-07-13:
Hello! For faster support, please contact our Discord chat: https://discord.gg/bpmXfz5
Meanwhile, here some tips:
* For the stream, you can try to tune EDID: https://docs.pikvm.org/edid
* For the KB/Mouse - do you see the keyboard/mouse icon on the top right corner of the web ui? Is it orange?

You can answer here, but I recommend duplicating your question and diagnostic answers in the chat. Our support will help you.

NalertN, 2023-07-14:
Hello
I tried EDID Examples from the site and after that it worked perfectly.
I used the EDID from the V2+ with the 1280x1024 resolution.

Thank you very much!

closed 2023-07-14 by NalertN

mdevaev, 2023-07-14:
Ur welcome!