pikvm/pikvm #1056: USB-HDMI on the pi zero 2 w

manasmgkar, 2023-07-14:
Pi zero 2 w is considerably faster than the original zero. 

I am asking this because where i live the cost of the hdmi-csi2 bridge is 2.5-4x of the PI itself. i cannot order from aliexpress and similar due to bans because of the political situation between china and my country. USB-HDMI is the cheapest and most practical solution for me.

i want to understand why a usb hub connected to the microusb on the pi zero with a usb-hdmi plugged into it wont work ? or a gpio based usb hub hat like this https://www.waveshare.com/usb-hub-hat-b.htm support he USB-HDMI dongle. 

or would the pi4 which again is expensive just to run pikvm be my only options

mdevaev, 2023-07-14:
* There is no GPIO-based USB HUB. If you look closely, you will see that this hub has spring-loaded contacts that touch the Zero board in the place where it has USB contact pads. Thus, this is a regular USB HUB, which, by the way, excludes the possibility of using a USB connector at Zero: although it is not physically occupied, the USB electrical connection is being made, so nothing else can be connected to it, so as not to damage all the devices involved in the connection.
* USB can operate in two mutually exclusive modes: host and device. To emulate the keyboard, mouse, and other devices, switching to device is required. After that, the host mode cannot be used, and the hub can only work with host.

manasmgkar, 2023-07-14:
> * There is no GPIO-based USB HUB. If you look closely, you will see that this hub has spring-loaded contacts that touch the Zero board in the place where it has USB contact pads. Thus, this is a [...]

makes a lot of sense. i am going to cancel my pi zero order then would just get the pi4